Arksine/moonraker #374: Turn on power when a print has been started

SimWalh, 2022-02-19:
### Is your feature request related to a problem? Please describe

I am using the [power] plugin to switch an ATX power supply for my printers 12V.
I have "locked_while_printing" set to True to prevent accidental power loss while printing, but this means i am not able to turn on the psu from within my start gcode via a macro:

[gcode_macro POWER_ON_PRINTER]
gcode:
  {action_call_remote_method("set_device_power",
                             device="printer",
                             state="on")}

### Describe the solution you'd like

Similar to "on_when_upload_queued", there should be an option to automatically turn the power on whenever a print is started, regardless if the file has been uploaded earlier. 
Maybe even replace the upload trigger with the print start trigger, as it makes more sense, since files could be uploaded without starting the print immediateley.

### Describe alternatives you've considered

Setting "locked_while_printing" to false and turning the printer on via the start gcode, but this also enables accidental shutdowns, via a misclick on a power off macro for example.

SimWalh, 2022-02-22:
It is possible, i'm doing it 😄 
i'm running a rumba board that is usb powered, and i am only switching 12v power for fans, motors, heaters etc., so klipper stays ready and connected.
Currently i am using the alternative solution from my initial post, wich works fine too, but is more risky.
I'm not sure what you mean by client? I'm using Mainsail as a web ui if that helps.

Arksine, 2022-02-22:
Your hardware configuration is very unique, I can't implement something in Moonraker that will not work well on most printers.  However, as I said, I could implement this with the queue which would work for anyone.

Yes, Mainsail is a client and it does support the queue.

ReubenBTalbott, 2022-04-08:
My printer is actually setup the same way, I have the BTT E3 Mini, all the 24V stuff is powered by a relay and the MCU is kept alive via USB power.

ibishop100, 2022-05-13:
I just want my LEDs to turn on when I start printing, even if I choose a job from previously uploaded ones.  Maybe I'm missing something, but with this:

[power enclosure_lights]
type: gpio
pin: gpio26
on_when_job_queued: True
off_when_shutdown: True
off_when_shutdown_delay: 300

I do not get lights unless I turn them on manually (if I choose a previously uploaded file).

mattheys, 2022-11-14:
I have a stock Ender 3 Pro and the board is powered by USB and I was using Octoprint fine to turn on the PSU for the Fans, Heater, Motor etc.  I can work around it by queueing the item first then printing it via the Mainsail interface but it's a bit counter intuative and it would be nice if there was another option to turn on the printer when a job starts, maybe a separate on_when_job_starts option?

synfinatic, 2022-12-22:
Going to +1 this request.   Just spent 4 hours debugging my printer with strange errors like:

```
TMC 'stepper_z' reports READRSP@RDSEL2: 00ffffff stallguard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1 se=31 sg_result@rdsel2=31
```

all because I forgot to turn on the printer power supply which drives the extruders.  Problem was caused because my Rasping is powered via a different PS than the extruders and the RasPi powers my Duet via USB.

khseal, 2022-12-23:
It would be nice to turn on the printer at the start of printing, as is done in octoprint. For many people, the logic and the power part are fed separately.

Arksine, 2022-12-24:
I took another look at this and have pushed a solution that should accommodate the requests here.  The `set_device_power` remote method now takes an optional `force` argument that can be used to toggle a device that is otherwise lock.  The documentation [provides an example](https://moonraker.readthedocs.io/en/latest/configuration/#toggling-device-state-from-klipper) of how to force on a device in a `PRINT_START` macro, effectively your Klipper configuration could add something like the following:
```
# printer.cfg

[gcode_macro POWER_ON_HEATERS]
gcode:
  {action_call_remote_method("set_device_power",
                             device="heaters",
                             state="on",
                             force=True)}

[gcode_macro PRINT_START]
gcode:
  # Turn on power supply for extruders/bed
  POWER_ON_HEATERS
  # Add a bit of delay to give the switch time
  G4 P2000
  # Add the rest of your "Start G-Code"...
```

This solution provides accidental off protection for the device, as the external APIs cannot be used to force toggle state.  They could potentially call your macro, however it would be possible to add a parameter to apply additional protection.  That said, if the macro only forces the device on it would have no affect on a print as the device is already on.

Additionally this solution works if the print is started outside of Moonraker, such as via the display.

khseal, 2022-12-24:
Strange, but with the force=True option, the code does not work.

khseal, 2022-12-24:
Everything is fine, but with drivers in uart mode, these macros will not work.
Error Unable to read tmc uart 'stepper_x' register IFCNT
It looks like the print needs to be enabled a little earlier than loading the code.

Arksine, 2022-12-25:
Its necessary to make sure your drivers have power and are initialized before they are enabled, otherwise the attempts at communicating over UART will fail.  The timing of this is going to depend on your hardware and configuration.  Your slicer may also need additional configuration.

khseal, 2022-12-25:
When I turn on the power manually before loading gcode, then everything is ok.

Arksine, 2022-12-25:
You may need to introduce additional delay after powering on the device.  This suggests that either the TMC drivers are not ready when enabled or interference after powering on is causing communication issues.  AFAIK nothing happens with the drivers when a print is started, only after the steppers are enabled do they start querying state.

khseal, 2022-12-26:
Turned the delay. Earned without errors.

Arksine, 2022-12-26:
Ok.  I should have thought of this before, but it is possible to power on your device before the print starts by overriding `SDCARD_PRINT_FILE`.  It isn't even necessary to use the `force` option here as the device power will be toggled before the print starts.

```ini
# printer.cfg

# Create a Macro to Power on the Heaters.  This is necessary to be
# sure that the template evaluates the call in the correct order.
[gcode_macro POWER_ON_HEATERS]
gcode:
  {action_call_remote_method("set_device_power",
                             device="heaters",
                             state="on")}

# Override SDCARD_PRINT_FILE
[gcode_macro SDCARD_PRINT_FILE]
rename_existing: SDCPF
gcode:
   # Step 1: Call the remote method to turn on the power device
   POWER_ON_HEATERS
   # Step 2: Pause while the device powers up.  The following example
   # pauses for 4 seconds.  It may be necessary to tweak this value.
   G4 P4000
   # Step 3: Call the renamed command to start the print
   SDCPF {rawparams}

```
As with above, you may need to adjust the delay.

khseal, 2022-12-26:
I checked your macros. Now new error:  "SD busy" =)

Arksine, 2022-12-26:
I can't reproduce.  There is likely something wrong with your setup, it sounds like `SDCARD_PRINT_FILE` is being called twice. Make sure you aren't calling `SDCARD_PRINT_FILE` from within the gcode file, nor should you be calling it from any other macros.

khseal, 2022-12-26:
I looked at my files and didn't find any duplicates.
Here is a sample code 
[Sushi_Press_Frame_30.zip](https://github.com/Arksine/moonraker/files/10304353/Sushi_Press_Frame_30.zip)

Arksine, 2022-12-26:
`SDCARD_PRINT_FILE` is in that gcode file on line 305.  It cannot be in the file.

khseal, 2022-12-26:
My mistake. I thought, need to write SDCARD_PRINT_FILE manually in start gcode. Everything works without errors.
Thanks for the power management update.

khseal, 2023-04-04:
After klipper updates or other updates.
Error on autostartup =(
Unable to read tmc uart 'stepper_x' register IFCNT
Change G4 P4000 no help...

HeedfulCrayon, 2023-09-15:
I am having the same issue as @KWall24 where it does not power on the printer on file upload or job queued
```
[power homeassistant_switch]
type: homeassistant
protocol: https
address: nunya
port: 443
device: switch.3d_printer_power
token: nunya
domain: switch
on_when_job_queued: True
restart_klipper_when_powered: True
```

KWall24, 2023-10-16:
@HeedfulCrayon I've since learned that `on_when_job_queued` refers to the Job Queue, which is separate from, in my case, telling PrusaSlicer to Upload and Print something. I can confirm, if I add something to the job queue, my printer powers on, so the function works as expected.

That being said, I'd love to find a way to make that process work, so that I can tell my slicer to Upload and Print, and the printer turns on and prints.

Arksine, 2023-10-16:
The `[file_manager]` section has to be [configured](https://moonraker.readthedocs.io/en/latest/configuration/#file_manager) to queue uploads when Klippy is not available by adding the `queue_gcode_uploads: True` option.  The [power documentation](https://moonraker.readthedocs.io/en/latest/configuration/#power-on-g-code-uploads) also contains details on the configuration requirements to power on the printer using "Upload and Print".

KWall24, 2023-10-16:
> The `[file_manager]` section has to be [configured](https://moonraker.readthedocs.io/en/latest/configuration/#file_manager) to queue uploads when Klippy is not available by adding the `queue_gcode_uploads: True` option. 

That [file_manager] bit was what I was missing. Thank you. I think that gets me where I need to go, in combination with `load_on_startup: True` within `[job_queue]`

Arksine, 2023-10-16:
Good to hear.  I think I can close this issue now, as the original request can be fulfilled through Klipper macros, and subsequent questions have been answered.

closed 2023-10-16 by Arksine

HeedfulCrayon, 2023-10-17:
I have this in my moonraker.conf and it still won't turn on automatically on upload
```
[file_manager]
enable_object_processing: False
queue_gcode_uploads: True

[job_queue]
load_on_startup: True
automatic_transition: False

[power homeassistant_switch]
type: homeassistant
protocol: https
address: nunya
port: 443
device: switch.3d_printer_power
token: nunya
domain: switch
on_when_job_queued: True
restart_klipper_when_powered: True
```

gretel, 2024-04-28:
> I have this in my moonraker.conf and it still won't turn on automatically on upload
> `on_when_job_queued: True`

same here.

Arksine, 2024-04-28:
The upload request has to set the print flag to `true` in order to be queued.

gretel, 2024-04-28:
> The upload request has to set the print flag to `true` in order to be queued.

yes, the print would start, if the printer did turn on which it does not hence my comment on here.

Arksine, 2024-04-28:
Two things can prevent the printer from powering on in this scenario:

- The print flag is not set
- The switch is not in the `off` state.  If its in an error condition or not initialized then it can't power on.

If you [enable verbose logging](https://moonraker.readthedocs.io/en/latest/installation/#the-environment-file) and reproduce the issue I'll take a look at the log.

gretel, 2024-04-28:
>     * The switch is not in the `off` state.  If its in an error condition or not initialized then it can't power on.

thanks, that was the actual root cause:
```
initial_state: off
```

is required, at least for me 👍🏼 

suggest to change the [according documentation](https://moonraker.readthedocs.io/en/latest/configuration/#power-on-g-code-uploads) to have the line in question not being commented out - think it will hurt less than [folks](https://www.reddit.com/r/OrcaSlicer/comments/1c4s666/moonraker_power_device_support/) running into this issue.

Arksine, 2024-04-28:
That generally shouldn't be necessary.  What it indicates is that the switch was likely turned off externally.  Only the `mqtt` power devices can reflect the state of external power changes.  Moonraker would have to poll the state for http devices to accomplish this, and even this is subject to timing issues.   Polling may be something I look at in the future, however it would be relatively far down the list of items I have to work on.

That said, I can make a small change that will attempt to refresh the device state when it receives the job queued event.  This would in effect turn on the device if it reports that its off.

gretel, 2024-04-28:
> That generally shouldn't be necessary. What it indicates is that the switch was likely turned off externally. Only the `mqtt` power devices can reflect the state of external power changes. Moonraker [...]

it's "just" an esphome device exporting a switch to `homeassistant`, so state keeping is plain vanilla - and nothing i can do much about but deviate from the defaults.

> That said, I can make a small change that will attempt to refresh the device state when it receives the job queued event. This would in effect turn on the device if it reports that its off.

should do the trick in terms of leaving other things as they are and documented?

Arksine, 2024-04-28:
> t's "just" an esphome device exporting a switch to homeassistant, so state keeping is plain vanilla - and nothing i can do much about but deviate from the defaults.

IIRC esphome devices support mqtt, but I'm not sure.  Regardless, if you  turn the switch off in Home Assistant then Moonraker won't know and its internal state will become stale.  Thus it will still think its on.

> should do the trick in terms of leaving other things as they are and documented?

It should.  Rather than check the internal state to see if its already on, Moonraker will request the state from homeassistant first.  If it reports off, it will then turn it on.

gretel, 2024-04-28:
> IIRC esphome devices support mqtt, but I'm not sure. Regardless, if you turn the switch off in Home Assistant then Moonraker won't know and its internal state will become stale. Thus it will still think its on.

`mqtt` got sort of superseded by using the state keeping of `homeassistant` solely. as you mentioned, polling is the way to go.
 
> > should do the trick in terms of leaving other things as they are and documented?
> It should. Rather than check the internal state to see if its already on, Moonraker will request the state from homeassistant first. If it reports off, it will then turn it on.

please let me validate 😏

Arksine, 2024-04-28:
Its pushed, so you can check it if you would like.  To give it a good test, pull the latest code, remove the `initial_state` option, restart the Moonraker service, power the switch on, then toggle the switch off from Home Assistant.   Moonraker's internal state will be stale, but it should still power on when you upload.

brucetony, 2024-12-19:
> After klipper updates or other updates. Error on autostartup =( Unable to read tmc uart 'stepper_x' register IFCNT Change G4 P4000 no help...
> 
> Updade. It seems I have solved the problem. I added to the start gcode G4 P2000

You, my friend, just saved me a lot of frustration. I knew I had to introduce delay somewhere to allow time for the stepper motors to initialize, but was too lazy to search for the marlin gcode command. Adding a G4 step to my `POWER_ON_PRINTER` macro was all that was needed, thanks!