Arksine/moonraker #379: unable to write json to moonraker DB API properly

ChokunPlayZ, 2022-02-25:
### What happened

I'm trying to migrate my print stat, the time, filament used, and other things from marlin to moonraker DB

I try to use the database API, it works but when I make an API call with JSON data the system escape all JSON strings so after I try to make an API call and see if it works this shows up on the response
```json
"history": "{\"job_totals\":{\"total_jobs\":652,\"total_time\":3743040.750717356,\"total_print_time\":3728791.0383284763,\"total_filament_used\":777772.91634,\"longest_job\":200649,\"longest_print\":200649}}"
```
and the frontend UI won't read data from it, I also tried getting the .mdb file off the raspberry pi and writing to it directly but I'm unable to access it on other devices it's only accessible on the pi
on other devices it will throw a "not an IMDB file" error

### Client

Other

### Browser

Other or N/A

### How to reproduce

1. make an HTTP POST Request to Moonraker Database API adding json data to the database
2. make another HTTP GET Request to Moonraker Database API requesting the data that just got written
3. from the result you will see that every quote is escaped and the data show up as a string, not JSON

Arksine, 2022-02-25:
That history namespace is protected and read only via the API.  There is no method to support migrating data from other firmwares.

With regard to the "not an lmdb file" error, its because lmdb files are architecture dependent.  You can't transfer them from a Pi to an x86 machine and directly read them.  Also, it is a very bad idea to modify the database directly.

ChokunPlayZ, 2022-02-25:
I enabled the debug mode in the config and the namespace become writable, the data is writing to the correct namespace, see picture
[image]
but the data being stored is not treated like json and being written to the database as a string

I also tried setting the content-type header to "application/json" and it has the same result

Arksine, 2022-02-25:
The debug mode is not intended for users.  Since you have enabled it and attempted to modify read-only portions the database there isn't much I can do in the way of assistance.  If you can reproduce the issue with a pristine database and provide `moonraker.log` I'm willing to look further at it.  In my tests the database works as expected.

ChokunPlayZ, 2022-02-25:
everything database related works as intended for me, the problem is the code treats the data as a string, not json data so it just writes string to the database, I wanted it to write JSON data to the database

here's the log from the last restart
[moonraker(1).log](https://github.com/Arksine/moonraker/files/8142630/moonraker.1.log)

Arksine, 2022-02-25:
All items in the database are stored as bytestrings.  They are encoded before storage and decoded after reading.  If you are getting a string back as the response, then they are not encoded correctly in the database.

SSH into your host and run the following commands:
```
cd ~
sudo apt install lmdb-utils
mdb_dump -p -s history ~/.moonraker_database -f history.db
```

This is going to create a `history.db` file in your home folder.  Get that file and upload it.

ChokunPlayZ, 2022-02-25:
got the file, the json that is not complete is my fault I'll get that fixed, scroll down for the complete ones
[history.db.txt](https://github.com/Arksine/moonraker/files/8143052/history.db.txt)

Arksine, 2022-02-25:
If you look at the file you will notice that all of the items are prefixed with an "s".  This means they are encoded as string values, not objects.  Your marlin migrator is not sending the correct value type with the POST request.  If you are using the query string the `value` field needs to use a type hint.  If you are sending via the body you need to make sure the `value` field is not enclosed in quotes.

FWIW, I am going to deprecate and remove the database debug option.  I understand why you want to do this, however this option was intended for Moonraker dev testing only, not as a means to modify write protected parts of the database.

ChokunPlayZ, 2022-02-25:
I didn't realize that I can use body, switching to the body make everything shows up as an object
 issue fixed, closing

closed 2022-02-25 by ChokunPlayZ