Arksine/moonraker #382: Allow re-ordering of job queue entries

garyd9, 2022-02-26:
### Is your feature request related to a problem? Please describe

The current job queue implementation appears to be a static FIFO list.  It would be more useful if items could be re-ordered.  For example, I might have 5 jobs pending in the queue, but realize that the 5th item should be printed before some others.  

### Describe the solution you'd like

Ideally, moonraker would support taking any item in the queue and moving it before any other item in the queue. 

### Describe alternatives you've considered

As a simple alternative, just the ability to move any item in the queue to the top of the queue (instead of to a selected position) would still be useful for those "oh, I need to print this next" type situations.

### Additional information

My perception of moonraker is only as seen via mainsail.  I apologize if this makes my descriptions seem odd.

pedrolamas, 2022-12-13:
I've been working on the [Job Queue support for Fluidd](https://github.com/fluidd-core/fluidd/pull/448), and I added re-order capability to it by clearing and re-adding all the jobs back into the queue, so it functionally it works, but I can't say I like it and would prefer a proper solution here...

I see two ways to improve this from the Moonraker side:

1. Add an optional boolean `replace_current` parameter to the [Enqueue a Job](https://moonraker.readthedocs.io/en/latest/web_api/#enqueue-a-job) operation, that when set to true, clears the existing job queue before adding the filenames (alternatively, make this a string param and pass "replace" or "append", with append being the default if ommited)
3. Add a new operation (`PATCH` perhaps?) that would allow passing in a `job_id` and its new index (or the full list of `job_id`s property ordered)

I have a working prototype of Moonraker changes for option 1 above, though option 2 is not bad either...

Would really like to hear your thoughts on this @Arksine, before I send any PR in 🙂

Arksine, 2022-12-22:
I looked at this and added a `/server/job_queue/jump` API since as its a relatively simple addition.  The endpoint takes a `job_id` parameter and will move the selected job to the front of the queue.

pedrolamas, 2022-12-22:
Thanks @Arksine, but I'm not sure we will be using this new endpoint as I have a feeling we will get users asking for the whole "why can't I order as I want?"...

My current implementation works by clearing the whole queue and adding the items back again in the new order, though it has some bad refresh side-effect

Here's an example of it working (and doing the `delete_job` and `post_job` on each interaction):

[image]

Arksine, 2022-12-22:
For a full reorder that is probably the best bet.  You could implement an "edit mode" that doesn't submit the request until the user is finished.  Then do a batch delete and queue.  This would also prevent unintentionally reordering.

pedrolamas, 2022-12-22:
That's an excellent idea!!

I do wonder if a `post_job` with a `replace_current` type of parameter (option 1 on the above) would still make sense, as it would mean we could do a single operation!

Arksine, 2022-12-22:
I would be ok with adding a "replace_current" or "reset" boolean parameter to the post request.

PR #574 (2022-12-23): job_queue: add "reset" parameter to post endpoint
Adds a new optional `reset` Boolean parameter to the POST endpoint of the `job_queue` component, allowing to clear the job queue before adding the new filenames.

Resolves #382
closed 2022-12-23 by Arksine