Arksine/moonraker #383: Repeated power on requests cause moonraker to restart Klipper repeatedly

ressu, 2022-02-26:
### What happened

KlipperScreen has a bug where it spams moonraker with power on requests (https://github.com/jordanruthe/KlipperScreen/issues/521). This causes moonraker to repeatedly restart Klipper due to `restart_klipper_when_powered` setting. 

```
[power printer]
restart_klipper_when_powered: True 
```

While the origin of the bug is in KlipperScreen, this is contrary to what documentation is leading to believe with "if Klippy reports that it is "ready", the  FIRMWARE_RESTART will be aborted as unnecessary". Klipper does report printer as ready, but `FIRMWARE_RESTART` is still invoked.

### Client

KlipperScreen

### Browser

Other or N/A

### How to reproduce

Set up power configuration in moonraker. Setting restart delay seems to make the issue more visible.

```
[power printer]
restart_klipper_when_powered: True 
restart_delay: 5
```

Power off the printer and use KlipperScreen splash screen to start the printer, you will observe multiple power on requests in the KlipperScreen log:

```
2022-02-26 22:17:21,905 [KlippyWebsocket.py:power_device_on()] - Sending machine.device_power.on printer
2022-02-26 22:17:21,906 [KlippyWebsocket.py:power_device_on()] - Sending machine.device_power.on printer
2022-02-26 22:17:21,907 [KlippyWebsocket.py:power_device_on()] - Sending machine.device_power.on printer
2022-02-26 22:17:21,908 [KlippyWebsocket.py:power_device_on()] - Sending machine.device_power.on printer
```

Alternatively it should be possible to send the power on requests in fast succession through the websocket and get a similar result.

Finally observe the repeated `FIRMWARE_RESTART` requests in moonraker console. Also observe the repeated Klipper ready messages:

```
22:19:06 // Klipper state: Ready
22:19:07 $ FIRMWARE_RESTART
22:19:07 $ FIRMWARE_RESTART
22:19:07 $ FIRMWARE_RESTART
22:19:11 // Klipper state: Ready
22:19:12 $ FIRMWARE_RESTART
```

### Additional information

The relevant parts of the logs are:

```
2022-02-26 22:16:27,758 [moonraker.py:add_log_rollover_item()] - 
#################### Moonraker Configuration ####################

[server]
host = 127.0.0.1
enable_debug_logging = False
klippy_uds_address = /run/klippy/uds

[authorization]
trusted_clients = 
        192.168.1.0/24
        10.55.0.0/16
        127.0.0.1
cors_domains = 
        https://my.mainsail.xyz
        https://app.fluidd.xyz
        https://nginx
force_logins = True

[octoprint_compat]

[history]

[file_manager]
queue_gcode_uploads = True
config_path = ~/printer_config

[job_queue]
load_on_startup = True

[power printer]
type = homeassistant
locked_while_printing = True
restart_klipper_when_powered = True
on_when_upload_queued = True
restart_delay = 5
address = x.x.x.x
device = switch.ressu3d_socket_1
token = xxxxxxxxxxxxxx

#################################################################
2022-02-26 22:16:27,761 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /server/info
2022-02-26 22:16:27,762 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.info
2022-02-26 22:16:27,762 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /server/config
2022-02-26 22:16:27,762 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.config
2022-02-26 22:16:27,762 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /server/restart
2022-02-26 22:16:27,762 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.restart
2022-02-26 22:16:27,764 [moonraker.py:load_component()] - Component (dbus_manager) loaded
[..]
System Time: 1645913812.202907, Usage: 2.08%, Memory: 48096 kB
System Time: 1645913813.204194, Usage: 2.25%, Memory: 48096 kB
System Time: 1645913814.202923, Usage: 2.16%, Memory: 48096 kB
System Time: 1645913815.205612, Usage: 2.13%, Memory: 48096 kB
CPU Temperature: 60.374
2022-02-26 22:16:56,074 [proc_stats.py:_handle_shutdown()] - Throttled Flags: 
2022-02-26 22:17:30,391 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2022-02-26 22:17:30,414 [websockets.py:build_error()] - JSON-RPC Request Error: 503
Klippy Host not connected
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 316, in func
    result = await self.klippy.request(
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/klippy_connection.py", line 405, in request
    raise ServerError("Klippy Host not connected", 503)
utils.ServerError: Klippy Host not connected
2022-02-26 22:17:31,656 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2022-02-26 22:17:31,912 [klippy_connection.py:_init_klippy_connection()] - Webhooks Subscribed
2022-02-26 22:17:31,916 [klippy_connection.py:_init_klippy_connection()] - GCode Output Subscribed
2022-02-26 22:17:36,195 [klippy_connection.py:_check_ready()] - Klippy ready
2022-02-26 22:17:36,681 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'temperature_sensor raspberry_pi', 'extruder']
2022-02-26 22:17:37,010 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2022-02-26 22:17:37,728 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2022-02-26 22:17:37,747 [websockets.py:build_error()] - JSON-RPC Request Error: 503
Klippy Host not connected
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 316, in func
    result = await self.klippy.request(
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/klippy_connection.py", line 405, in request
    raise ServerError("Klippy Host not connected", 503)
utils.ServerError: Klippy Host not connected
2022-02-26 22:17:38,994 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2022-02-26 22:17:39,250 [klippy_connection.py:_init_klippy_connection()] - Webhooks Subscribed
2022-02-26 22:17:39,259 [klippy_connection.py:_init_klippy_connection()] - GCode Output Subscribed
2022-02-26 22:17:41,524 [klippy_connection.py:_check_ready()] - Klippy ready
2022-02-26 22:17:42,013 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'temperature_sensor raspberry_pi', 'extruder']
2022-02-26 22:17:42,315 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2022-02-26 22:17:43,050 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2022-02-26 22:17:43,078 [websockets.py:build_error()] - JSON-RPC Request Error: 503
Klippy Host not connected
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 316, in func
    result = await self.klippy.request(
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/klippy_connection.py", line 405, in request
    raise ServerError("Klippy Host not connected", 503)
utils.ServerError: Klippy Host not connected
2022-02-26 22:17:44,313 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2022-02-26 22:17:44,567 [klippy_connection.py:_init_klippy_connection()] - Webhooks Subscribed
2022-02-26 22:17:44,569 [klippy_connection.py:_init_klippy_connection()] - GCode Output Subscribed
2022-02-26 22:17:46,844 [klippy_connection.py:_check_ready()] - Klippy ready
2022-02-26 22:17:47,328 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'temperature_sensor raspberry_pi', 'extruder']
2022-02-26 22:17:47,644 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2022-02-26 22:17:48,373 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2022-02-26 22:17:48,392 [websockets.py:build_error()] - JSON-RPC Request Error: 503
Klippy Host not connected
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py", line 316, in func
    result = await self.klippy.request(
  File "/mnt/dietpi_userdata/octoprint/moonraker/moonraker/klippy_connection.py", line 405, in request
    raise ServerError("Klippy Host not connected", 503)
utils.ServerError: Klippy Host not connected
2022-02-26 22:17:49,634 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2022-02-26 22:17:49,893 [klippy_connection.py:_init_klippy_connection()] - Webhooks Subscribed
```

Arksine, 2022-02-26:
> While the origin of the bug is in KlipperScreen, this is contrary to what documentation is leading to believe with if Klippy reports that it is "ready", the FIRMWARE_RESTART will be aborted as unnecessary. Klipper does report printer as ready, but FIRMWARE_RESTART is still invoked.

The full documentation states
```
If it isn't possible to schedule a firmware restart (ie: Klippy is disconnected), the restart
will be postponed until Klippy reconnects and reports that startup is  complete.  In this scenario,
if Klippy reports that it is "ready", the  FIRMWARE_RESTART will be aborted as unnecessary.
```
Notice that this scenario does not apply here, as Klippy is connected when the power on commands are sent.  That said the frequency with which these commands are being sent is sneaking past the lock I use to prevent concurrent requests.   I need to move the lock up a level in the code in a way that doesn't block requests for other devices to prevent this from occurring.

Arksine, 2022-02-27:
This should now be resolved on Moonraker's side.  A lock will prevent reentry before the state is queried, blocking subsequent requests until the current one is finished.  These requests should abort as the device will report as "on" when they are executed.

That said, it would be a good idea for KlipperScreen to fix this, as isn't ideal to spam any API request.

ressu, 2022-02-27:
I can confirm that the fix works. Thank you.

closed 2022-02-27 by ressu