Arksine/moonraker #384: Unparsed errors upon start

Dennisd80, 2022-02-27:
### What happened

After updating to the most recent version of moonraker(moonraker
v0.7.1-439-gf93fed6) I get some weird unparsed errors:

[moonraker.py:load_component()] - Component (power) loaded
2022-02-27 19:50:06,411 [moonraker.py:add_warning()] - Unparsed config option 'config_path: ~/klipper_config' detected in section [server].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.
2022-02-27 19:50:06,411 [moonraker.py:add_warning()] - Unparsed config option 'log_path: ~/klipper_logs' detected in section [server].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.
2022-02-27 19:50:06,411 [moonraker.py:add_warning()] - Unparsed config option 'machine_name: Ender3' detected in section [server].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.
2022-02-27 19:50:06,411 [moonraker.py:add_warning()] - Unparsed config option 'restart_delay: 1.' detected in section [power ledstrip].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.

The moonraker configuration file was not touched.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Restarting

Arksine, 2022-02-27:
You have old, deprecated options in your configuration.  Fluidd recently added the ability to show warnings which is why you are just noticing this, these messages have been there for a while.  The solution is to remove the deprecated options.

Arksine, 2022-03-16:
Since this seems to be resolved I'm going to close this.  Thanks.

closed 2022-03-16 by Arksine