Arksine/moonraker #389: After Moonraker update,My webUI can not connect MY

Terbinm, 2022-02-28:
### What happened

After I updated my moonraker, the web page doesn't seem to be able to connect to klipper. I tried waiting, restarting, and this post: https://github.com/fluidd-core/fluidd/issues/186. But can't solve my problem
[image]
[moonraker.log](https://github.com/Arksine/moonraker/files/8153930/moonraker.log)
[image]

This is my first time asking a question on github, please let me know if I'm missing something

### Client

Fluidd

### Browser

Microsoft Edge

### How to reproduce

Update Moonraker, jump out and restart after the update is complete,

### Additional information

[image]
[moonraker.log](https://github.com/Arksine/moonraker/files/8153930/moonraker.log)
[image]

Davidq656, 2022-02-28:
I just updated moonraker and got the same issue niw.

Arksine, 2022-02-28:
@Davidq656 its likely that you are missing deps too.  Run the install script as mentioned above, logging in as the normal user.  You likely also need to run `~/moonraker/scripts/set-policykit-rules.sh` afterward, the install script will warn you if the polkit rules aren't detected.

Terbinm, 2022-03-01:
@Arksine  The system can read my UI config new...but it can not connect to the Klipper either.

[image]

Terbinm, 2022-03-01:
@Arksine I try to cover moonraker.conf with WinSCP, and it did!
 maybe the conf had been changing? I guess

Arksine, 2022-03-01:
The configuration hasn't changed in a way that would prevent the server from starting.  Most likely the config was corrupted or incorrectly formatted, resulting in a parsing error on startup.  The log at `moonraker.log` would provide an indication as to what happened, its always important to include it when you have an issue.

Terbinm, 2022-03-01:
thank a lot.

Arksine, 2022-03-16:
Since this seems to be resolve I am going to close this issue.  Thanks.