Arksine/moonraker #392: Progress option in notifications

TheLion, 2022-03-06:
### Is your feature request related to a problem? Please describe

No

### Describe the solution you'd like

The new notifier option is great and works great. 

Is it possible to create some kind of progress options well?

So it sends a status and screenshot every x min or layers. 

### Describe alternatives you've considered

Te ram plugin, but I use Pushover for all notifications, so through Apprise should be great!

thankyousam, 2024-02-16:
I know this is a couple of years old now, but I've been thinking of a similar feature, so looking for anyone other folks ideas on this.

I have been looking at the existing code which generates the events that trigger the existing notifier actions. There is a stub of code in job_state.py which generates a layer_change event, but nothing in notifier currently listens for it. 

There is a limited amount of data available in the existing print state objects, and even less when the layer change event occurs. - it's pretty much the current layer, total layer, and a job and print duration, and not even the file name of the currently printing file. There's also nothing around the ETA, or progress through the gcode.

The obvious and easy way, which I've quickly coded locally, is to provide a progress notification based on how many layers have been completed against the total layer count, e.g. trigger every 25% during the printing. I've created a new notifier event which can be added to the moonraker.conf and the % trigger supplied as config.

Another option would be to provide a progress event after every X layers, rather than every Y percent, but again this is based on the layer progress, not the true runtime of the overall print. The good thing about the percent is you can essentially say how many progress updates you want during the job, i.e. 25% will always provide 3 updates (25%, 50% and 75% - 0% is start and 100% is complete so no need to trigger). The every X layers would provide a regular heartbeat, but would obvs be variable based on the number of layers in the job.

I've put some code together which I'll look to add into a branch for review.

TheLion, 2024-02-24:
Thanks, would be great!
Now using Telegram for progress and pushover for starts/stops ;-)

thankyousam, 2024-04-12:
PR #840 created to enable this feature in Notifier.