Arksine/moonraker #396: Overlapped paths

adelyser, 2022-03-13:
### What happened

With the updates to moonraker, I now get this error:
`Failed to register 'config': '/home/pi', path overlaps reserved path 'moonraker': '/home/pi/moonraker'`

I know it's because I have my configs directory set to my home directory, instead of a sub-directory, which I prefer and have always setup this way. 

There should be a method to disable this check if someone wanted to use their home directory for their configs instead of forcing us to use a configs directory that doesn't overlap with the moonraker directory.

### Client

Fluidd

### Browser

Brave

### How to reproduce

Use the latest moonraker and set your config_path to your home directory

Arksine, 2022-03-13:
This was a security vulnerability that was recently patched, it allowed for Moonraker's source code to be overwritten.  The only way to grand r/w access to the home folder would be to move Moonraker, Klipper, and Moonraker's database out of it.

Unfortunately it isn't possible to make this configurable as it would reintroduce the vulnerability for everyone.

Arksine, 2022-03-13:
After some thought, it should be possible to allow the configuration and reject write operations in protected folders, this is something I'll look into.  It would still be an error for the gcode and config paths to overlap though, as the inotify watches would overlap

cchance27, 2022-04-20:
is there a fix for this as this seems to occur when we use kiauh to setup things since the latest update?

Arksine, 2022-04-22:
As of right now the fix is to make sure that the configured paths do not overlap.  A standard installation, of which KIAUH should perform, shouldn't result in this issue.  I would suggest asking for assistance with KIAUH on the Klipper discord.

After the next beta release I plan to change how paths are configured.  I am moving them out of the configuration and into the command line.  When this is done I'll relax some of the restrictions on what is allowed to be configured, users will be able to get creative with symbolic links and serve different areas of the file system.  Note that I still don't recommend serving your entire home directory, I can't promise that it won't cause an issue, but it will be possible to do it.

Arksine, 2022-11-19:
FWIW, with the new changes overlapped paths should no longer be an issue.  While it isn't recommended it would be possible to serve the entire home directory using symbolic links in the data path.

As mentioned above, there still could be potential issues as this isn't something its designed for and I haven't personally tested it.  If you really want to serve the entire home directory it would probably be best to put the data path in `/opt` somewhere, making sure Moonraker has the correct permissions.

closed 2022-11-19 by Arksine