Arksine/moonraker #397: Moonraker Warning - Supplied path for gcodes

midhunmonachan, 2022-03-15:
### What happened

I'm getting two warnings on my dashboard and they are the same.

It says 
```
Supplied path (/home/pi/sdcard) for (gcodes) is invalid. Make sure that the path exists and is not the file system root.
```

I'm not sure where its getting that path from. In my cfg file I have 
```
[virtual_sdcard]
path: ~/gcode_files
```

I couldn't find anything about this error anywhere. But I'm able to upload gcode files from mainsailos file manager and they end up in `~/gcode_files` in my pi

Here is the log:

```
--------------------Moonraker Log Start--------------------
config_file: /home/pi/klipper_config/moonraker.conf
log_file: /home/pi/klipper_logs/moonraker.log
software_version: v0.7.1-453-gee312ee
python_version: 3.7.3 (default, Jan 22 2021, 20:04:44)  [GCC 8.3.0]

#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
enable_debug_logging = False
max_upload_size = 1024

[file_manager]
config_path = ~/klipper_config
log_path = ~/klipper_logs
enable_object_processing = False

[authorization]
cors_domains = 
	https://my.mainsail.xyz
	http://my.mainsail.xyz
	http://*.local
	http://*.lan
trusted_clients = 
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128

[octoprint_compat]

[history]

[update_manager]
refresh_interval = 168

[update_manager mainsail]
type = web
repo = mainsail-crew/mainsail
path = ~/mainsail

#################################################################
Unsafe Shutdown Count: 7

System Info:
[... 218 lines cut ...]
  File "/home/pi/moonraker/moonraker/klippy_connection.py", line 423, in request
    return await self._request_standard(web_request)
  File "/home/pi/moonraker/moonraker/klippy_connection.py", line 478, in _request_standard
    return await base_request.wait()
  File "/home/pi/moonraker/moonraker/klippy_connection.py", line 548, in wait
    raise self.response
utils.ServerError: Klippy Disconnected
2022-03-14 23:16:25,319 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2022-03-14 23:16:25,572 [klippy_connection.py:_init_klippy_connection()] - Webhooks Subscribed
2022-03-14 23:16:25,573 [klippy_connection.py:_init_klippy_connection()] - GCode Output Subscribed
2022-03-14 23:16:27,389 [klippy_connection.py:_check_ready()] - Klippy ready
2022-03-14 23:16:27,879 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'extruder']
```

### Client

Mainsail

### Browser

Firefox

### How to reproduce

Not really sure. It was there since the beginning of my install

Arksine, 2022-03-15:
At some point you configured klipper's `[virtual_sdcard]` to use the path `/home/pi/sdcard`.  The warning was generated because the path does not exist.  You reconfigured to `/home/pi/gcode_files` which got it working.  To clear the warning you simply need to restart Moonraker.

Arksine, 2022-04-22:
As this discussion seems to have concluded I am going to close this.

closed 2022-04-22 by Arksine

Brick935, 2024-02-24:
I ran into the same issue lately, and I am unsure on how to restart Moonraker. It's probably a dumb question but I am I still struggling. Any help would be greatly appreciated