Arksine/moonraker #399: Moonraker crash after attempted update

safik01, 2022-03-16:
### What happened

Hi, 
I doubt this is a real bug, but I ran out of ideas and I would appreciate any recomendation what to try next...

Everything worked as usual and I checked for updates using mainsail  as I do every once in a while. Updates for KlipperScreen and three system packages where available. I clicked Upgrade everything and got **Connection failed** - Cannot connect to Moonraker.

I waited a few minutes and still couldn't connect. I did a power off-on cycle and both KlippeScreen and Mainsail still unable to connect. (KlipperScreen shows splash screen with _Connecting to printer_ message)

I updated the system packages via ssh, Moonraker is still dead.
I pulled Moonraker from git and installed  `./install-moonraker.sh -f -c /home/pi/printer_config/moonraker.conf -r`. This ended alright with message `###### Launching Moonraker API Server...` but nothing can connect.

I am not able to identify anything suspicius in the log except for the fact that the most recent log entries are much shorter than the entry from before.

### Client

Mainsail, KlipperScreen

### Browser

Chrome

### How to reproduce

I have no idea..

### Additional information

The log file has over 3MB now so here is just a snippet:
[moonraker__.log](https://github.com/Arksine/moonraker/files/8262339/moonraker__.log)

Mainsail error log is full of: 
`2022/03/16 08:03:17 [error] 643#643: *43 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.1.40, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.1.28"`

DragonicDefson, 2022-10-17:
Can confirm, got the same problem, but just on one single PI, still investigating, one PI doesn't crash, one does. They are both the exact same setup for 2 printers.

Still an issue. Mainsail unable to connect to Moonraker.

DragonicDefson, 2022-10-17:
The exception I got is a little different, it seems that Moonraker doesn't close it's connections the correct way, this is on a local network, so it seems a bit off the scale to see it's full worker connection pool being filled up.

Will follow up my with my investigation, dig a little deeper.

2022/10/17 02:22:01 [alert] 532#532: *1092 768 worker_connections are not enough while connecting to upstream, client: 127.0.0.1, server: _, request: "GET /websocket?token= HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket?token=".

Arksine, 2022-10-17:
The OP appeared to have an corrupt instance, likely a file system issue that resulted in an incomplete merge after the update.  There are a large number of unsafe shutdowns detected in his log which can lead to file system and repo corruption.

I'll need `moonraker.log` to diagnose your issue @DragonicDefson.  You appear to have posted a snippet from another log, presumably one generated by nginx.  FWIW, nginx is responsible for managing its connection pool, not moonraker.

If the instance is corrupt, the most complete way to correct the issue is to remove the repo and re-clone via ssh:
```
sudo service moonraker stop
cd ~
rm -rf moonraker
git clone https://github.com/arksine/moonraker.git
sudo service moonraker start
```

DragonicDefson, 2022-10-17:
I'll take a look into it right now, give me a minute or 10.

Arksine, 2022-10-17:
Ok.  If Moonraker was unable to restart after the update it will likely go through the process of validating the install.  You may be required to enter your sudo password, I presume you would have already gone through this with your other Pi.

DragonicDefson, 2022-10-17:
Per accident maybe? But it's strange because it shuts down occasionally, it runs for a few hour straight with no issues, and suddenly shuts down.

DragonicDefson, 2022-10-17:
Have looked into it, the NGINX error.log literally contains or lines, the same applies to the access.log file.

DragonicDefson, 2022-10-17:
Contains 0 lines, apologies it's getting late here.

Arksine, 2022-10-17:
Where is that snippet coming from, the browser?  It should connect to nginx, therefore there should be something in `mainsail-access.log` (presuming you are on MainsailOS)

DragonicDefson, 2022-10-17:
That snippet is indeed coming from mainsail-access.log, it's also the only thing it reports, there is no additional warning or exception logged that gives us a clue where the problem comes from.

DragonicDefson, 2022-10-17:
Will try what you stated above, remove it, re-clone it, will let you know tomorrow afternoon after work if the problem is resolved.

DragonicDefson, 2022-10-18:
Hey! Just got back from work, and the issue is indeed resolved after re-deploying it like you stated.

Have to say, big shout out to you regarding the latest Moonraker update, it even runs with more performance right now than it did before on the RPI 1 B+

Apparently I watched over that whole sudo password notification, and since I run UFW on it didn't allow port 7125/tcp.
If I can make a small suggestion, let it report that same link inside the console as well.

Arksine, 2022-10-18:
It appears that Klippy is not starting, `klippy.log` may provide some context for that. 

With regard to SSH, that is unrelated to Moonraker.  Presumably something went wrong with the OpenSSH server.  You can try restarting the PI (see if Moonraker can reboot it through Mainsail before pulling the plug).

DragonicDefson, 2022-10-18:
I'm really sure it's an issue with Moonraker itself, this is the error it displays now, and still on that one single PI.

[image]

Arksine, 2022-10-18:
An error with Moonraker would not prevent you from connecting over SSH.  The error in that screenshot suggests Moonraker has an issue starting, however the previous one suggested that Klippy had an issue starting.  Were you able to reboot the Pi and connect via SSH?

If I had to guess, you have a file system issue that is not recoverable without reformatting and reinstalling.  You may be able to recover some individual items if you can connect via ssh/scp.

DragonicDefson, 2022-10-18:
Can be the case indeed if I'm thinking about it now, the PI that gives those issues has a bit more degraded performance compared to the other PI.

DragonicDefson, 2022-10-18:
Will try to reformat it, I've compiled-in support for H264 hardware transcoding on both of the PI's but luckily I got an backup of those binaries left on the still running PI. That will at least spare me some time.

Arksine, 2022-11-11:
As discussion seems to have concluded I will close this.  If the issue persists feel free to reopen.

closed 2022-11-11 by Arksine

DragonicDefson, 2022-11-11:
Yes, resolved, dead SD card, ordered new ones, already installed.

Apologies for no response, thanks for the assistance.