Arksine/moonraker #402: Moonraker Notifier: Attached webcam snapshot has no file extesion and isnt recognized as .jpg

Agent772, 2022-03-21:
### What happened

When using the notifier function of moonraker and attaching a Webcam Snapshot via URL (as shown in the example config) does not show a image. 
The attached file has no extension and therefore isnt recognized as jpg (picture).

Here the moonraker.conf:
`[notifier print_finished]
url: https://discord.com/api/webhooks/......
events: complete
body: Print completed **{event_args[1].filename}**
  Print Time: **{(event_args[1].total_duration / 3600)|round(0,'floor')|int}:{((event_args[1].total_duration / 60)|round(0) -((event_args[1].total_duration / 3600)|round(0,'floor')*60))|round(0)|int}**
**attach: http://127.0.0.1/webcam/?action=snapshot**`

This results in:
[image]

after download and renaming of the webcam file to `webcam.jpg` it it contains the expected picture

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Problem occures every time the complete event is triggered

Agent772, 2022-03-21:
Thanks for the quick awnser. Sry i thought i uploaded the log, updated my inital post with it.
I tried your tip with the name but no success. 
Il look into the documentation of apprise and the camera software on the pi

Arksine, 2022-03-21:
Thanks.  After looking at the log, it looks like you need to slightly modify your URL since the request already has a query string:
```
[notifier print_finished]
attach = http://127.0.0.1/webcam/?action=snapshot&name=webcam.jpg
```

As you can see I replaced the second `?` with a `&`.  Since your log contained a token in the URL I remove it from the OP.

Agent772, 2022-03-21:
Thanks that solved it.
Maybe you could mention that in the example part of the config documentation

PS: The Token was randomized so all was good :)

closed 2022-03-21 by Agent772