Arksine/moonraker #403: Tasmota power stopped working

grantemsley, 2022-03-24:
### What happened

Upgraded from Moonraker commit 9d7baa1 to ac8c440.

Since upgrading my tasmota power control is not working. It shows up in mainsail, but clicking it doesn't work. I think it's because the command being sent to tasmota has some extra symbols in it.

### Moonraker power config
```
[power printer]
type: tasmota
off_when_shutdown: False
on_when_job_queued: True
restart_klipper_when_powered: True
restart_delay: 5
address: 192.168.5.187
output_id: 1
```

I tried adding the new output_id, but it didn't seem to make a difference.
I turned on debug logging so I can see exactly what's going on and found this:

```
2022-03-24 13:32:21,136 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"machine.device_power.off","params":{"printer":null},"id":7188}
2022-03-24 13:32:21,138 [curl_httpclient.py:_curl_setup_request()] - GET http://192.168.5.187/cm?user=admin&password=&cmnd=Power1
2022-03-24 13:32:21,140 [curl_httpclient.py:_curl_debug()] - Hostname 192.168.5.187 was found in DNS cache
2022-03-24 13:32:21,140 [curl_httpclient.py:_curl_debug()] - Trying 192.168.5.187:80...
2022-03-24 13:32:21,145 [curl_httpclient.py:_curl_debug()] - Connected to 192.168.5.187 (192.168.5.187) port 80 (#17)
2022-03-24 13:32:21,146 [curl_httpclient.py:_curl_debug()] - > GET /cm?user=admin&password=&cmnd=Power1 HTTP/1.1
2022-03-24 13:32:21,146 [curl_httpclient.py:_curl_debug()] - > Host: 192.168.5.187
2022-03-24 13:32:21,147 [curl_httpclient.py:_curl_debug()] - > User-Agent: Moonraker
2022-03-24 13:32:21,147 [curl_httpclient.py:_curl_debug()] - > Accept: */*
2022-03-24 13:32:21,147 [curl_httpclient.py:_curl_debug()] - > Accept-Encoding: gzip,deflate
2022-03-24 13:32:21,147 [curl_httpclient.py:_curl_debug()] - >
2022-03-24 13:32:21,160 [curl_httpclient.py:_curl_debug()] - Mark bundle as not supporting multiuse
2022-03-24 13:32:21,160 [curl_httpclient.py:_curl_debug()] - < HTTP/1.1 200 OK
2022-03-24 13:32:21,160 [curl_httpclient.py:_curl_debug()] - < Content-Type: application/json
2022-03-24 13:32:21,161 [curl_httpclient.py:_curl_debug()] - < Server: Tasmota/9.5.0 (ESP8266EX)
2022-03-24 13:32:21,161 [curl_httpclient.py:_curl_debug()] - < Cache-Control: no-cache, no-store, must-revalidate
2022-03-24 13:32:21,161 [curl_httpclient.py:_curl_debug()] - < Pragma: no-cache
2022-03-24 13:32:21,162 [curl_httpclient.py:_curl_debug()] - < Expires: -1
2022-03-24 13:32:21,162 [curl_httpclient.py:_curl_debug()] - < Accept-Ranges: none
2022-03-24 13:32:21,162 [curl_httpclient.py:_curl_debug()] - < Transfer-Encoding: chunked
2022-03-24 13:32:21,162 [curl_httpclient.py:_curl_debug()] - < Connection: close
2022-03-24 13:32:21,163 [curl_httpclient.py:_curl_debug()] - <
2022-03-24 13:32:21,175 [curl_httpclient.py:_curl_debug()] - Closing connection 17
2022-03-24 13:32:21,248 [curl_httpclient.py:_curl_setup_request()] - GET http://192.168.5.187/cm?user=admin&password=&cmnd=Power1%2520off
2022-03-24 13:32:21,250 [curl_httpclient.py:_curl_debug()] - Hostname 192.168.5.187 was found in DNS cache
2022-03-24 13:32:21,251 [curl_httpclient.py:_curl_debug()] - Trying 192.168.5.187:80...
2022-03-24 13:32:21,255 [curl_httpclient.py:_curl_debug()] - Connected to 192.168.5.187 (192.168.5.187) port 80 (#18)
2022-03-24 13:32:21,256 [curl_httpclient.py:_curl_debug()] - > GET /cm?user=admin&password=&cmnd=Power1%2520off HTTP/1.1
2022-03-24 13:32:21,256 [curl_httpclient.py:_curl_debug()] - > Host: 192.168.5.187
2022-03-24 13:32:21,256 [curl_httpclient.py:_curl_debug()] - > User-Agent: Moonraker
2022-03-24 13:32:21,257 [curl_httpclient.py:_curl_debug()] - > Accept: */*
2022-03-24 13:32:21,257 [curl_httpclient.py:_curl_debug()] - > Accept-Encoding: gzip,deflate
2022-03-24 13:32:21,257 [curl_httpclient.py:_curl_debug()] - >
2022-03-24 13:32:21,271 [curl_httpclient.py:_curl_debug()] - Mark bundle as not supporting multiuse
2022-03-24 13:32:21,272 [curl_httpclient.py:_curl_debug()] - < HTTP/1.1 200 OK
2022-03-24 13:32:21,273 [curl_httpclient.py:_curl_debug()] - < Content-Type: application/json
2022-03-24 13:32:21,273 [curl_httpclient.py:_curl_debug()] - < Server: Tasmota/9.5.0 (ESP8266EX)
2022-03-24 13:32:21,274 [curl_httpclient.py:_curl_debug()] - < Cache-Control: no-cache, no-store, must-revalidate
2022-03-24 13:32:21,274 [curl_httpclient.py:_curl_debug()] - < Pragma: no-cache
2022-03-24 13:32:21,274 [curl_httpclient.py:_curl_debug()] - < Expires: -1
2022-03-24 13:32:21,274 [curl_httpclient.py:_curl_debug()] - < Accept-Ranges: none
2022-03-24 13:32:21,275 [curl_httpclient.py:_curl_debug()] - < Transfer-Encoding: chunked
2022-03-24 13:32:21,275 [curl_httpclient.py:_curl_debug()] - < Connection: close
2022-03-24 13:32:21,276 [curl_httpclient.py:_curl_debug()] - <
2022-03-24 13:32:21,280 [curl_httpclient.py:_curl_debug()] - Closing connection 18
2022-03-24 13:32:21,290 [power.py:_schedule_firmware_restart()] - Klipper reports 'ready', aborting FIRMWARE_RESTART
```

The first curl to the tasmota device looks fine, querying it's status. The second one however has cmnd set to `Power1%2520off` instead of `Power1%20off`. I'm not sure where that 25 came from, but it makes the command invalid and so nothing happens.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Setup a tasmota device and try to toggle power

Arksine, 2022-03-24:
Thanks.  You are correct, I just pushed a series of commits that escape all power urls as some users were reporting tokens with special characters.  I didn't realize that the Tasmota implementation hardcoded some escaped characters in the url.  This should be fixed in 1bdedd009f0550047870bf10cbf82a0c3a5c85df.

grantemsley, 2022-03-24:
Just updated and confirmed it is working as expected now.

Thanks for the quick fix!

closed 2022-03-24 by grantemsley