Arksine/moonraker #404: Alternate form of job enqueue doesn't work

ressu, 2022-03-27:
### What happened

Job queue API documentation describes that there are 2 forms of calls to the enqueue API. Either by using request parameters of a JSON payload. https://moonraker.readthedocs.io/en/latest/web_api/#enqueue-a-job

Sending a request with JSON body results in an error:

```
POST /server/job_queue/job HTTP/1.1
Accept-Encoding: identity                                                                                                                                              
Content-Length: 48
Content-Type: applicaton/json                                                                                                                                          

{
  "filenames": [
    "test/dummy.gcode",
  ]
}
```

```
{"error": {"code": 400, "message": "Bad Request", "traceback": "Traceback (most recent call last):\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 640, in _process_http_request\n    result = await self._do_request(args, conn)\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 616, in _do_local_request\n    return await self.callback(\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/components/job_queue.py\", line 249, in _handle_job_request\n    files: Union[List[str], str] = web_request.get('filenames')\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py\", line 121, in get\n    raise ServerError(f\"No data for argument: {key}\")\n\nutils.ServerError: No data for argument: filenames\n\n\nThe above exception was the direct cause of the following exception:\n\n\nTraceback (most recent call last):\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1704, in _execute\n    result = await result\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 606, in post\n    await self._process_http_request()\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 642, in _process_http_request\n    raise tornado.web.HTTPError(\n\ntornado.web.HTTPError: HTTP 400: Bad Request (No data for argument: filenames)\n"}}
```

Which makes sense since the code seems to be assuming that the request will always be a request parameter:

https://github.com/Arksine/moonraker/blob/384ab71d729cf75c5471d1ac4783a2a6c0a399d3/moonraker/components/job_queue.py#L248-L253

### Client

Other

### Browser

Other or N/A

### How to reproduce

Send a request as follows:

```
POST /server/job_queue/job HTTP/1.1
Accept-Encoding: identity                                                                                                                                              
Content-Length: 48
Content-Type: applicaton/json                                                                                                                                          

{
  "filenames": [
    "test/dummy.gcode",
  ]
}
```

The test above is assuming that the dummy.gcode exists, but shouldn't affect outcome even if it doesn't:

```
GET /server/files/metadata?filename=test%2Fdummy.gcode HTTP/1.1
Accept-Encoding: identity                                                                                                                                              
Content-Type: applicaton/json
```

```
{
  "result": {
    "filename": "test/dummy.gcode",
    "gcode_end_byte": 64,
    "gcode_start_byte": 37,  
    "job_id": null,
    "modified": 1648382340.422621,
    "print_start_time": null,
    "size": 64, 
    "slicer": "Unknown"
  }
}
```

Arksine, 2022-03-27:
Moonraker parses arguments from both the query string and a json body.  I suspect the issue is a malformed JSON object:
```
{
  "filenames": [
    "test/dummy.gcode",
  ]
}
```
There shouldn't be a comma at the end.  This appears to be an error in Moonraker's documentation as well, which I will fix.

ressu, 2022-03-27:
Just tested without the comma:

```
POST /server/job_queue/job HTTP/1.1
Accept-Encoding: identity                                                                                                                                              
Content-Length: 47
User-Agent: vial-http
Content-Type: applicaton/json                                                                                                                                          
Host: printer.192-168-1-76.sslip.io

{
  "filenames": [
    "test/dummy.gcode"
  ]
}
```

Result is the same:

```
{"error": {"code": 400, "message": "Bad Request", "traceback": "Traceback (most recent call last):\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 640, in _process_http_request\n    result = await self._do_request(args, conn)\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 616, in _do_local_request\n    return await self.callback(\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/components/job_queue.py\", line 249, in _handle_job_request\n    files: Union[List[str], str] = web_request.get('filenames')\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/websockets.py\", line 121, in get\n    raise ServerError(f\"No data for argument: {key}\")\n\nutils.ServerError: No data for argument: filenames\n\n\nThe above exception was the direct cause of the following exception:\n\n\nTraceback (most recent call last):\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1704, in _execute\n    result = await result\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 606, in post\n    await self._process_http_request()\n\n  File \"/mnt/dietpi_userdata/octoprint/moonraker/moonraker/app.py\", line 642, in _process_http_request\n    raise tornado.web.HTTPError(\n\ntornado.web.HTTPError: HTTP 400: Bad Request (No data for argument: filenames)\n"}}
```

Arksine, 2022-03-27:
Upload the full log.  There is something about your request that is resulting in a failure to parse the body.

ressu, 2022-03-27:
[moonraker (3).log](https://github.com/Arksine/moonraker/files/8357660/moonraker.3.log)

Log attached

ressu, 2022-03-27:
Sigh.. I think I see it, `Content-Type: applicaton/json`. Note the missing `i`

Ok, disregard. This issue (number 404) was indeed not found :D

closed 2022-03-27 by ressu