Arksine/moonraker #405: websockets.py:build_error() - json-rpc request error method not found

henryrgithub, 2022-03-29:
### What happened

Hello,
[moonraker.log](https://github.com/Arksine/moonraker/files/8367107/moonraker.log)

I'm getting this error repeatedly. Fluidd shows unable to connect to klippy, and I think this is why. Is there a quick solution such as manually pulling the websockets library or some such?

Thanks!

### Client

Fluidd

### Browser

Firefox

### How to reproduce

Install raspberry pi os 64bit. Install klipper manually. Install Moonraker and Fluidd via Kiauh script.

Arksine, 2022-03-29:
There is an issue with Klipper.  Since you did a manual install the most likely cause is that you did not enable [Klipper's API server](https://moonraker.readthedocs.io/en/latest/installation/#enabling-the-unix-socket).

If that is not the cause then there is an error preventing Klipper from starting.

henryrgithub, 2022-03-29:
Yes that was it! Thanks for the heads up. Mine was set up to use systemd as the docs say it might be, so I navigated to /etc/systemd/system and edited klipper.service, adding [-a /tmp/klippy_uds] (ignore brackets) to the end of the ExecStart line. Restarted my pi and it worked.

Thanks again for your help!

closed 2022-03-29 by henryrgithub