Arksine/moonraker #408: Moonraker update giving policy errors, advice from #349 not fixing.

ssjbardock, 2022-04-03:
### What happened

Having issues with my moonraker giving warnings on Fluidd after using the built in update button, similar to #349, but the advice to run set-policykit-rules.sh isn't working for me. 

Advised by @arksine to make a new issue.

I've tried deleting the existing policy, as well as reinstalling moonraker, but still just getting the same errors
[image]
[image]
.

[moonraker(2).log](https://github.com/Arksine/moonraker/files/8404829/moonraker.2.log)

Using a Pi 3B+, was working normally for months before I updated to the newest Moonraker.

I'm still able to print on Fluidd fine, but the warnings are there constantly.

### Client

Fluidd

### Browser

Firefox

### How to reproduce

Updated using built in update manager. After updating Moonraker, errors appeared, and won't leave regardless of methods tried.

Arksine, 2022-04-03:
Thanks.  One thing I notice immediately, the hostname in the log `fluiddpi` does not match the hostname of the screenshot you provided running `set-policykit-rules` ('raspberrypi`).  Are you sure you ran the script on the right host?

ssjbardock, 2022-04-03:
I'm new to all this. I was under the impression the 'host' was the raspberry pi that Fluidd/Moonraker/Klipper was installed on, which I SSH'd into and ran the scrips from the #349 link.

Is that not the case?

ssjbardock, 2022-04-03:
AH! I see what my mistake may have been. Didn't understand that fluiddpi was a separate part I can also SSH into, which explains why I couldn't find the Moonraker in the pi SSH the first time and had to reinstall.

Trying the command again in correct fluiddpi SSH.

Yup that did it. My misunderstanding, issue closed!

closed 2022-04-03 by ssjbardock