Arksine/moonraker #411: [server] elements reported as invalid

haplm, 2022-04-04:
### What happened

I'm getting following errors after update to v0.7.1-508-g9c33eb2:

```2022-04-04 22:00:34,944 [moonraker.py:add_warning()] - Unparsed config option 'config_path: ~/klipper_config' detected in section [server].  This may be an option no longer available or could be the result of a module that failed to lo
ad.  In the future this will result in a startup error.
2022-04-04 22:00:34,944 [moonraker.py:add_warning()] - Unparsed config option 'temperature_store_size: 600' detected in section [server].  This may be an option no longer available or could be the result of a module that failed to load
.  In the future this will result in a startup error.
2022-04-04 22:00:34,944 [moonraker.py:add_warning()] - Unparsed config option 'gcode_store_size: 1000' detected in section [server].  This may be an option no longer available or could be the result of a module that failed to load.  In
 the future this will result in a startup error.
2022-04-04 22:00:34,944 [moonraker.py:add_warning()] - Unparsed config option 'log_path: ~/klipper_logs' detected in section [server].  This may be an option no longer available or could be the result of a module that failed to load.  
In the future this will result in a startup error.```

I can't find any information about them being removed

### Client

Mainsail

### Browser

Microsoft Edge

### How to reproduce

Update to the latest version, restart

Arksine, 2022-04-04:
They were moved [last November](https://moonraker.readthedocs.io/en/latest/user_changes/#november-7th-2021), however I allowed configuration from both until today's push.  The [configuration docs](https://moonraker.readthedocs.io/en/latest/configuration/) have details.  In your case the store sizes should be in `[data_store]` and the paths should be in `[file_manager]`,

haplm, 2022-04-04:
Ah, sorry, when I was looking at the doc, I was checking https://moonraker.readthedocs.io/en/stable/configuration/ because that's what google returns as first result (as opposed to https://moonraker.readthedocs.io/en/latest/configuration/#server)

Thanks, and sorry!

closed 2022-04-04 by haplm