Arksine/moonraker #415: TLS support for MQTT

RealDeuce, 2022-04-09:
### Is your feature request related to a problem? Please describe

MQTT sends userID and password in the clear unless TLS is used, moonraker MQTT doesn't appear to support TLS

### Describe the solution you'd like

A configuration option to enable TLS and optionally define ca_certs should be all that's needed, then before the endpoint is registered, calling tls_set() on the paho_mqtt object (with ca_certs if configured) should be all that's needed.

### Additional information

This appears to work:

```diff
diff --git a/moonraker/components/mqtt.py b/moonraker/components/mqtt.py
index 4650cf3..89cf2fc 100644
--- a/moonraker/components/mqtt.py
+++ b/moonraker/components/mqtt.py
@@ -190,6 +190,8 @@ class MQTTClient(APITransport, Subscribable):
         self.subscribed_topics: SubscribedDict = {}
         self.pending_responses: List[asyncio.Future] = []
         self.pending_acks: Dict[int, asyncio.Future] = {}
+        if self.port == 8883:
+            self.client.tls_set(ca_certs="/home/admin/my-ca.crt")
 
         self.server.register_endpoint(
             "/server/mqtt/publish", ["POST"],
```

Arksine, 2024-04-28:
TLS support has been added with the `enable_tls` option.   That said, if you are using a self signed cert you need create your own CA and install the root certificate on the machine hosting Moonraker.

ilf-S, 2024-09-21:
I want to connect to a mosquitto server using x509 certs for encryption and client auth. I have my own CA. I glanced the code and it seems you can only use TLS to encrypt the server to client connection, but cannot use client certs. Any chance of this getting implemented, I know paho supports it.

RealDeuce, 2024-09-26:
Just a quick note, I do have a CA, but I don't want it trusted for anything except MQTT, and want it to be the only one Moonraker trusts an MQTT to be signed by.

There's really two issues here:
1) Setting SSL_CERT_DIR and/or SSL_CERT_FILE for the single process is a bit awkward and non-obvious.
2) I don't want MQTT to trust all the CAs that my browser does.

In writing this, I realized that if I only have the one CA available for Moonraker, all the update stuff will likely break.

I'm struggling to think of a case where someone would use TLS with their MQTT broker, but want to support all the CAs that are used for https.