Arksine/moonraker #416: Error when attempting to power on plug via HomeAssistant

dannyglover, 2022-04-11:
### What happened

My plug is listed in fluidd in the power section dropdown. I can toggle the power state successfully there, via clicking the button.
However, when attempting to get moonraker to turn on the plug automatically via the `on_when_job_queued` option, I get this:
Note: `192.168.1.30` is my HomeAssistant url.

`Plug Config:`
```
[file_manager]
config_path: /home/pi/klipper_config
log_path: /home/pi/klipper_logs
queue_gcode_uploads: True

[job_queue]
load_on_startup: True

[power tapo_plug_ender]
type: homeassistant
address: 192.168.1.30
port: 8123
device: switch.ender_3_creality
token: my_long_token_here
domain: switch
protocol: http
off_when_shutdown: True
off_when_shutdown_delay: 600
on_when_job_queued: True
restart_klipper_when_powered: True
restart_delay: 2
locked_while_printing: True
```

`Moonraker:`
```
2022-04-11 13:06:30,763 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.1.30
2022-04-11 13:06:31,022 [app.py:log_request()] - 101 GET /websocket?token=3YG722A4SEQNGXCTERN5A4AUKC63IJPC (192.168.1.30) [_TRUSTED_USER_] 5.80ms
2022-04-11 13:06:31,023 [websockets.py:open()] - Websocket Opened: ID: 548336165360, Proxied: True, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/100.0.4896.75 Safari/537.36, Host Name: octopi-ender3.station
2022-04-11 13:07:11,868 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/file_manager/metadata.py -p /home/pi/gcode_files -f "CE3_CHEPCalibrationCube.gcode") successfully finished
2022-04-11 13:07:11,878 [app.py:log_request()] - 201 POST /server/files/upload (192.168.1.30) [_TRUSTED_USER_] 944.70ms
2022-04-11 13:07:12,277 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Method not found
```

`HomeAssistant:`
```
2022-04-11 13:01:32 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 334, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
  File "aiohttp/_http_parser.pyx", line 551, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message="Bad status line 'Invalid method encountered'"
```

Am I doing something wrong, or is there an internal issue here?
Thank you for the great work on Moonraker and thank you for taking the time to read this report.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

- Use a tapo plug
- Attempt to turn the plug on via fluidd's power menu, notice the power turns on successfully.
- Try to turn on the plug when a job is queued, notice that it does not work

### Additional information

https://www.dropbox.com/s/bipru52ftwm1ugv/moonraker%20%2811%29.log?dl=0

Log was too big to attach. Uploaded to dropbox instead.

Arksine, 2022-04-11:
It appears that you have a gcode file that is timing out during metadata extraction.  Specifically, `preprocess-cancellation` is taking too long (over 5 minutes).  Unfortunately I'm not sure what file it is, I should add some logging for that.  I should probably also cap the size of objects that are allowed to be processed.  The timeout is resulting in an error, which is why the printer doesn't power on.

In the interim I would recommend disabling object processing in the `[file_manager]`. If you need are testing exclude object you could set up your slicer to do the processing, this is the best solution for large gcode files unless you are running Moonraker on desktop class hardware.  A Pi 4 with an SSD could probably handle anything you throw at it, unfortunately the slow I/O on anything else really gets bogged down and Moonraker won't wait indefinitely for it to finish.

dannyglover, 2022-04-11:
> It appears that you have a gcode file that is timing out during metadata extraction. Specifically, `preprocess-cancellation` is taking too long (over 5 minutes). Unfortunately I'm not sure what file it is, I should add some logging for that. I should probably also cap the size of objects that are allowed to be processed. The timeout is resulting in an error, which is why the printer doesn't power on.
> 
> In the interim I would recommend disabling object processing in the `[file_manager]`. If you are testing exclude object you could set up your slicer to do the processing, this is the best solution for large gcode files unless you are running Moonraker on desktop class hardware. A Pi 4 with an SSD could probably handle anything you throw at it, unfortunately the slow I/O on anything else really gets bogged down and Moonraker won't wait indefinitely for it to finish.

You were absolutely right! Uploading it straight from PrusaSlicer did result in it working. If it helps in your debugging, here is the gcode it was running: https://www.dropbox.com/s/mx7i3m65en79jiz/CE3_CHEPCalibrationCube.gcode?dl=0

It's just a basic calibration cube. I was testing the functionality via the upload button in Fluidd. I intend on using the PrusaSlicer auto upload and print workflow however. I was just using the Fluidd UI for testing, as I failed in my attempts to get any of this automation working previously, as I was missing some macros.

Thank you very much for all the great work you do on Moonraker!

Arksine, 2022-04-11:
Thanks, that is a small gcode, it definitely shouldn't have timed out.  I'll see if I can reproduce with with the object processor on my end, I might have to send this up the chain to the guys that maintain that library.

Arksine, 2022-04-13:
FWIW, I ran your gcode file through the parser and didn't have an issue.  I wonder if an older version of the `preprocess-cancellation` dependency is on your system?  If you ssh into the Pi and run the following command:
```
~/moonraker-env/bin/pip freeze
```
It will spit out all of the installed dependencies along with their versions.  If you could paste that here it would be helpful.  Thanks.

dannyglover, 2022-04-23:
> FWIW, I ran your gcode file through the parser and didn't have an issue. I wonder if an older version of the `preprocess-cancellation` dependency is on your system? If you ssh into the Pi and run [...]

Sorry for the long delay, had a car accident last week and have had tons to sort out as a result. 
Here's the requested info:

```
apprise==0.9.7
certifi==2021.10.8
charset-normalizer==2.0.12
click==8.1.2
dbus-next==0.2.3
distro==1.5.0
idna==3.3
ifaddr==0.1.7
importlib-metadata==4.11.3
inotify-simple==1.3.5
Jinja2==3.0.3
libnacl==1.7.2
lmdb==1.2.1
Markdown==3.3.6
MarkupSafe==2.1.1
oauthlib==3.2.0
paho-mqtt==1.5.1
Pillow==9.0.1
preprocess-cancellation==0.2.0
pycurl==7.44.1
pyserial==3.4
pyserial-asyncio==0.6
PyYAML==6.0
requests==2.27.1
requests-oauthlib==1.3.1
six==1.16.0
streaming-form-data==1.8.1
tornado==6.1
urllib3==1.26.9
zeroconf==0.37.0
zipp==3.7.0
```

Thanks again.

Arksine, 2022-07-09:
I wasn't ever able to reproduce this issue, it seems to have been a strange one off.  All of your dependencies look correct.  I'm going to go ahead and close this for now, however if the issue resurfaces feel free to reopen.

closed 2022-07-09 by Arksine