Arksine/moonraker #419: Cant connect to Fluidddpi

eggie326, 2022-04-13:
### What happened

 I get this error 

http://fluiddpi.local
No moonraker connection. Please check moonraker status and / or refresh.

after trying to fix this

    [server]: Option 'config_path' has been moved to section [file_manager]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation.
    [server]: Option 'log_path' has been moved to section [file_manager]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation.
    [server]: Option 'temperature_store_size' has been moved to section [data_store]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation.
    [server]: Option 'gcode_store_size' has been moved to section [data_store]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation 

Is there a way to fix this

### Client

Fluidd

### Browser

Firefox

### How to reproduce

NA

Arksine, 2022-04-13:
It sounds like the config file failed parsing.  You are going to have to ssh into your Pi, correct the configuration, and restart Moonraker.

The log should tell you exactly what went wrong.  I'm unsure as to where FluiddPi puts the log, it is either at `~/klipper_logs/moonraker.log` or `/tmp/moonraker.log`.

corrado-c, 2022-04-13:
he gave me a similar writing after updating.

`Moonraker warnings found.
[server]: Option 'config_path' has been moved to section [file_manager]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation.
[server]: Option 'log_path' has been moved to section [file_manager]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation.
[server]: Option 'temperature_store_size' has been moved to section [data_store]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation.
[server]: Option 'gcode_store_size' has been moved to section [data_store]. Please correct your configuration, see https://moonraker.readthedocs.io/en/latest/configuration for detailed documentation.`

iliquifysnacks, 2022-04-14:
I'd also be interested in how to resolve this. It's not that I changed anything about my configuration, it just failed to work after the update...

Arksine, 2022-04-14:
As mentioned above, the log files will provide details on why Moonraker failed to start.   You can use `scp` to transfer the log to your local machine.  If you are using FluiddPi or MainsailOS, the log is probably in the `klipper_logs` folder.

If you are on Windows, presumably PowerShell is installed and is recent enough to support scp:
```
scp pi@fluiddpi.local:~/klipper_logs/moonraker.log .
```
Replace `pi@fluiddpi.local` with the username@hostname of your printer's sbc/pc (presumably a raspberry Pi).  If the file isn't in `klipper_logs`, check `/tmp`:
```  
scp pi@fluiddpi.local:/tmp/moonraker.log .
```

If you don't want to use the command line (or can't because PowerShell doesn't support it), you can try [WinSCP](https://winscp.net/eng/index.php).

Once you have logs, attach them to this thread and we can determine the issue.

The warnings in this thread indicate that your are using a legacy configuration.  To resolve them move the options from the `[server]` section in to the specified area.  Looking at the post from @corrado-c, there are for options in the `[server]` section that need to be moved to resolve the warning.  Two must go in `[file_manager]` and two must go in `[data_store]`.  If those sections haven't been created you should create them yourselves.  Make sure that the section isn't duplicated, this will cause an error.  No whitespace before the section or before options.  If you are familiar with modifying Klipper's configuration, Moonraker's behaves in a similar fashion.

Both Mainsail and Fluidd provide a UI for updating `moonraker.conf`.  Make sure that the config file is formatted correctly or it could result in the OP's situation where Moonraker can't start.

iliquifysnacks, 2022-04-14:
Thank you very much, I moved the sections and cleared the errors.

corrado-c, 2022-04-15:
Hello!  I solved it too, I had some problems with the formatting I guess  , and I was wrong to restart only klipper ... only after I thought about restarting moonraker!

3vilson, 2022-04-15:
You need to change the **moonraker.conf** file from

```
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False
config_path: ~/klipper_config
temperature_store_size: 600
gcode_store_size: 1000
log_path: ~/klipper_logs
```

to this 

```
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False

[file_manager]
config_path: ~/klipper_config
log_path: ~/klipper_logs

[data_store]
temperature_store_size: 600
gcode_store_size: 1000
```

Ensiba, 2022-04-15:
> You need to change the **moonraker.conf** file from this
> 
> ```
> [server]
> host: 0.0.0.0
> port: 7125
> enable_debug_logging: False
> config_path: ~/klipper_config
> temperature_store_size: 600
> gcode_store_size: 1000
> log_path: ~/klipper_logs
> ```
> 
> to this
> 
> ```
> [server]
> host: 0.0.0.0
> port: 7125
> enable_debug_logging: False
> 
> [file_manager]
> config_path: ~/klipper_config
> log_path: ~/klipper_logs
> 
> [data_store]
> temperature_store_size: 600
> gcode_store_size: 1000
> ```
> 
> then reboot

Thanks A Lot! That worked perfectly!

Xcion54, 2022-04-17:
> You need to change the **moonraker.conf** file from this
>
> ```
> [server]
> host: 0.0.0.0
> port: 7125
> enable_debug_logging: False
> config_path: ~/klipper_config
> temperature_store_size: 600 [...]

Thanks for this, I dont understand why updating moornraker doesnt reformat this

Arksine, 2022-04-17:
Currently Moonraker has no mechanism to update the configuration itself.  That said, it is something I plan to add.  I don't intend to make any further breaking changes without a way to automate them if at all possible.

I know the config and polkit changes were frustrating. I plan to tag a beta soon, when that happens updates will occur at a slower pace unless you opt-in to the dev channel.     That is one of the goals of the beta channel, which will eventually lead to a stable release. Thanks for hanging in there.

corrado-c, 2022-04-17:
no problem!!
once I understand how to do it is simple!
Everything I have already done is fantastic !.

FlatEarthDave, 2022-04-18:
I don't know why the dev couldn't have just answered the question on his first reply. It's not like you didn't know those sections were changed, when the OP stated the configuration file error in his post. Someone else had to post the solution, which the dev  blamed on a "legacy" version, which must mean any version older than a week? I am still grateful for the API and all your hard work. Thanks for that.

Arksine, 2022-04-20:
@FlatEarthDave On the first reply I was responding to a separate issue.  The deprecations (and the solution to them) would not cause a failure to parse moonraker.conf. 

Legacy configuration means months old, going back to last November.  The issue is that a "new" install of FluiddPi is bundled with an old version of Moonraker and a legacy configuration.

kcmccombs, 2022-05-11:
okay i have done the edit from above but now i get a 501 error can someone help with this thanks

Creo2005, 2022-05-12:
> okay i have done the edit from above but now i get a 502 error can someone help with this thanks

The same error
[image]
:

Arksine, 2022-05-12:
With regard to the 502 errors, if you are attempting to access via `https://app.fluidd.xyz` on chrome then sadly it will no longer work.  This is a browser issue, you can still use Firefox as an alternative.

Otherwise I will need `moonraker.log`.  If you are on FluiddPi you should be able to get it with the following command...using either PowerShell on windows 10 or a terminal on MacOS/Linux:
```
scp pi@hostname:~/klipper_logs/moonraker.log .
```
The `hostname` should be the hostname or ip address of the Pi in question.  It will download the file to the working directory shown in the terminal.

Its difficult to provide assistance for different users in the same issue tracker.  It may appear that you have the same issue, but its possible the cause is different.  Since the original poster's issue has been resolved, I'm going to close this one.  If you continue to have a 502 and arent using the PWA (app.fluidd.xyz), please create a new issue using the issue template and attach moonraker.log.  Alternatively you can ask for help on the Klipper discord in the `moonraker` channel. I would recommend trying discord first as you may get a response more quickly.

closed 2022-05-12 by Arksine