Arksine/moonraker #423: Queued items don't get recorded in history

ressu, 2022-04-18:
### What happened

When queueing a print from SuperSlicer through the Octoprint API, when the printer is powered off, the print doesn't get recorded in history. When the printer is on, a newly submitted print will get recorded into history as usual.

### Client

Other

### Browser

Other or N/A

### How to reproduce

1. Configure printer to use a power plug via home assistant
2. Power off printer
3. Queue print from PrusaSlicer through the octoprint integration
4. Observe the print start and not appear in history
5. Wait for the print to complete or cancel print

In neither case the print appears in the history

### Additional information

The printer is controlled via HomeAssistant and is configured to automatically start on queue.

The log from my current print is here [moonraker (4).log](https://github.com/Arksine/moonraker/files/8508166/moonraker.4.log). An upload via the Octoprint API happened at 21:39. After I realised that I forgot to power on the printer, I used the emergency stop and FIRMWARE_RESTART before manually restarting the print from Mainsail.

Arksine, 2022-04-19:
I can't reproduce this behavior on my system, however I added some logging that will be helpful in tracking this down. 
 I also added a bit of additional delay to the start time of queued jobs.  My suspicion is there is a race condition due to timing and Moonraker isn't receiving a "printing" status update from Klipper on your system when a queued job is started after power on.

If you could pull the latest version of Moonraker and try to reproduce it would be helpful.  You can add `enable_debug_logging: True` to the `[server]` section to provide details.  Just remember to disable later as will fill up the log quickly.

Also, as a side note it appears that Home Assistant is requesting metadata on the thumbnail files.  I would suggest filtering those requests before sending them.

ressu, 2022-04-19:
I can confirm that the prints now appear in history as expected with the latest version from git (v0.7.1-478-g384ab71).

I'll also take a look at the metadata, that shouldn't be HomeAssistant, but I suspect that is Mooncord.. but I'll file the appropriate bugs to the proper locations after triaging. Thanks for the prompt fix!

closed 2022-04-19 by ressu