Arksine/moonraker #425: Allow binding to both IPv4 and IPv6

jvandenbroek, 2022-05-01:
### Is your feature request related to a problem? Please describe

Currently it's not possible, or at least I'm unable to figure it out, to listen on both IPv4 and IPv6. If you configure "host: ::" it will listen to IPv6 but not on IPv4 and localhost.

### Describe the solution you'd like

Allow to specify multiple hosts, or listen to all for both IPv4 and IPv6.

wlhlm, 2022-08-04:
I've hit this as well. Doing some digging, I found that the framework moonraker uses for networking [explicitly disables](https://www.tornadoweb.org/en/stable/_modules/tornado/netutil.html#bind_sockets) receiving IPv4 when listening on IPv6. That means you need to listen on both 0.0.0.0 and :: separately. I don't think moonraker supports multiple host directives.

There is a workaround: if you specify a hostname for the host directive, the framework will bind to all addresses to which the hostname resolves. For example `host: localhost` will result in moonraker listening on both ::1 and 127.0.0.1. So you can listen on the hostname of the host moonraker is running on and it should listen on all addresses (v4 and v6) from which the host is reachable, which should effectively work the same as listening on 0.0.0.0/:: (with the drawback being that moonraker won't pick up new addresses should they change after moonraker has been started).

Arksine, 2022-08-19:
This should work with commit 97c8b05eee0ca799659e490c9c883d4517946d7a, however I am not currently running a dual stack network so I cannot test it.  You can bind to all interfaces by specifying `host: all`  in the `[server]` section.  I chose to keep Tornado's default behavior so a user can still bind to only ipv6 interfaces if they so choose.

Sonicboom247, 2022-08-19:
does commit 97c8b05 address the notorious error when trying to access via IP?
403 Forbidden
nginx/1.18.0 (Ubuntu)

Arksine, 2022-08-19:
No, it doesn't address any errors.  Moonraker only raises a 403 in two scenarios:
1) A request attempts to access a forbidden namespace in the database
2) A request attempts to access a file that is locked by Klipper or otherwise not accessible.

Most likely nginx is responsible for that error, which suggests its configuration needs to be modified.

Sonicboom247, 2022-08-19:
You are correct sir, it is an ngix error but there is little info on how to correct it..

jvandenbroek, 2022-08-21:
> This should work with commit [97c8b05](https://github.com/Arksine/moonraker/commit/97c8b05eee0ca799659e490c9c883d4517946d7a), however I am not currently running a dual stack network so I cannot test it. You can bind to all interfaces by specifying `host: all` in the `[server]` section. I chose to keep Tornado's default behavior so a user can still bind to only ipv6 interfaces if they so choose.

I can confirm it works. Thanks :+1:

closed 2022-08-21 by jvandenbroek