Arksine/moonraker #431: HTTP 500 error while uploading print from Prusa or super slicer

ahmadmadary, 2022-05-13:
### What happened

Hi
I just have just updated the moonraker to v0.7.1-569-gbabfba1, and now I get the following error while trying to upload the print to the printer:
[image]

I tried to save the gcode and upload it to the printer from the mainsail interface, but print does not start.

can you point me in a direction to solve the issue?
Thanks

### Client

Mainsail

### Browser

Chrome

### How to reproduce

I update the klipper to v0.10.0-408-g2c441b45
and moonraker to v0.7.1-569-gbabfba1

Arksine, 2022-05-13:
Thanks.  This was a bug introduced in 09ac00ce2eb63519a3fd5d16af22543613f04f70.  It should be fixed by 0164bb239ea1032e105428435ff2643378bf4bfa.

ahmadmadary, 2022-05-13:
Thanks. I had to also reinstall the mainsail for some reason as if after updating the moonraker, mainsail was gone!
anyway, thank you for solving this.

closed 2022-05-13 by ahmadmadary