Arksine/moonraker #432: log_path not honoured

ilium007, 2022-05-14:
### What happened

klipper version: v0.10.0-408-g2c441b45

moonraker version: v0.7.1-570-g0164bb2

moonraker.conf:

```
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False

[file_manager]
config_path: ~/klipper_config
log_path: /var/log
```

log_path is not honoured, moonraker.log file still in ~/klipper_logs

```
pi@raspberrypi:~/klipper_logs $ ll
total 15756
drwxr-xr-x  2 pi pi    4096 May 14 02:08 .
drwxr-xr-x 19 pi pi    4096 May 14 04:03 ..
-rw-r--r--  1 pi pi 3196735 May 14 04:07 klippy.log
-rw-r--r--  1 pi pi 9629890 May 13 16:38 klippy.log.2022-05-13
lrwxrwxrwx  1 pi pi      34 May 13 11:23 mainsail-access.log -> /var/log/nginx/mainsail-access.log
lrwxrwxrwx  1 pi pi      33 May 13 11:23 mainsail-error.log -> /var/log/nginx/mainsail-error.log
-rw-r--r--  1 pi pi 1660363 May 14 04:07 moonraker.log
-rw-r--r--  1 pi pi 1612822 May 13 23:57 moonraker.log.2022-05-13
lrwxrwxrwx  1 pi pi      20 May 13 11:28 webcamd.log -> /var/log/webcamd.log
pi@raspberrypi:~/klipper_logs $
```

I am running log2ram to avoid SD cards writes on the Raspberry Pi

### Client

Mainsail

### Browser

Safari

### How to reproduce

Set log_path in moonraker.conf as above

### Additional information

Would also like to know how to move klippy.log to /var/log

ilium007, 2022-05-14:
Even here on the moonraker restart the path is wrong:

[image]

Arksine, 2022-05-14:
The `log_path` is a location where Moonraker serves a `logs` folder, it does not change the location where Moonraker's log is placed.  The log file is set on the command line.  The install script can be used to change its location, ie:
```
~/moonraker/scripts/install-moonraker.sh -f -c ~/klipper_config/moonraker.conf -l /var/log/moonraker.log
```

Just as an FYI, the in the future the `log_path` will be deprecated.  For security reasons I plan to remove all of the `*_path` options in Moonraker and put all of Moonraker's file dependencies in a single folder.  It would still be possible to create a symbolic link to `/var/logs` from this folder though.

ilium007, 2022-05-14:
I installed via kiuah, I'll have a look back there and see if there are any options. Thanks.

closed 2022-05-14 by ilium007