Arksine/moonraker #433: Moonraker will not start if TP link plug IP is incorrect

LazaroFilm, 2022-05-19:
### What happened

I use a TP Link as smart power plug. Unfortunately, with my TP link router, I cannot assign a fixed IP to smart plug (I know it's completely dum, I submitted a ticket to TP link and they replied "Why would you even want a fixed ip for your smart plug" /rant)
When the IP of the plug changes (like during our last power outage) Moonraker will not boot (Mainsail shows connection failed message).
The solution is to SSH into the pi via VS-Code and correct the IP address in `moonraker.conf` and reboot the Pi.

The preferred behavior would be to let Moonraker start and disable the plug until the IP address is changed back. That would allow me to access the config file through Mainsail instead of going through VS-Code

### Client

Mainsail

### Browser

Safari

### How to reproduce

To create the error:
- Enter a wrong IP address in the [power] section for a TP Link plug.
- Reboot the Pi.
- Open Mainsail in the browser and try to connect.

To fix it:
- Enter a valid IP address in `moonraker.conf`
- reboot Pi.

### Additional information

My [power] section in `moonraker.conf`
```
[power Printer]
type: tplink_smartplug
address: 123.456.78.910
off_when_shutdown: True
locked_while_printing: True
restart_klipper_when_powered: True
restart_delay: 3
```

Arksine, 2022-05-19:
Moonraker should still start if the IP address is invalid, however the power device will not be available.  I will need you to provide `moonraker.log` showing this particular behavior.  FWIW, it shouldn't be necessary to reboot the pi after changing the config, restarting the Moonraker service should suffice.

LazaroFilm, 2022-05-19:
Here is a [pastebin of the moonraker.log](https://pastebin.com/bk7xVaAf)
Yes I don't need to restart the whole pi, but `reboot` is faster to type.

LazaroFilm, 2022-05-19:
So, I tried entering manually a wrong PI and yes, indeed it reboots and shows the error. I will need to investigate deeper. Maybe it's when my router loses power and changes the IP of the plug? I'll investigate further. I'll close this until I can recreate the event more reliably. For now I'll know that If moonraker doesn't start for my setup, I should double check the IP of the plug.

closed 2022-05-19 by LazaroFilm