Arksine/moonraker #434: The ability to limit the rate MQTT messages.

alexrmay91, 2022-05-20:
### Is your feature request related to a problem? Please describe

When setting up objects to publish MQTT messages (specifically temperatures), messages are sent extremely rapidly and with a huge amount of decimals. Documentation states that object state is only sent if the value has changed, but with temperatures to so many decimal places, it will always change.

### Describe the solution you'd like

I would like to be able to limit the rate for specific objects. Something like the following:

     status_objects:
       webhooks
       toolhead=position,print_time,homed_axes
       extruder=temperature
         rate: 1000 #ms between mesages

Alternatively, a way to round floats to a smaller number of decimals may limit the rate at which messages get published.

### Describe alternatives you've considered

Using REST would be an alternative because I can specify the rate queries occur, but I prefer push communication to polling communication.

contagon, 2022-09-18:
I'd like to add a +1 for this feature. After anything more than a couple of minutes publishing, I'm getting thousands of messages which seems unnecessary.

dkneisz, 2024-04-27:
I'd like to push this topic. I want to include my printer in my openHAB with MQTT. Right now this is sending lots of messages, so it's unusable, because the data will also end up in a database. In Octoprint you have the option to set a threshold for the temperature. It would be nice to have at least that option here to to limit the messages sent. Currently I get around 4 to 5 messages for the various temperatures every second. In that time the temperature has changed by 0.02 °C. That is more or less useless information.