Arksine/moonraker #438: Temperature not editable on Mobile Device

SchwabJ, 2022-05-28:
### What happened

Hello,
Since the last Version i cant edit the temperatures on mobile devices:

[image]

Is this bug known? The previous verson works.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

E.g. Pixel 6 Pro Android 12 with Chrome

### Additional information

v0.7.1-572-gd53cf27

Creo2005, 2022-05-28:
> ### What happened
> Hello, Since the last Version i cant edit the temperatures on mobile devices:
>
> [image]
>
> Is this bug known? The previous verson works.
>
> ### Client
> Mainsail
>
> ### Browser
> Chrome
>
> ### [...]

Use Android app: Klipperoid or Mobileraker :)

SchwabJ, 2022-05-28:
> 
> Use Android app: Klipperoid or Mobileraker :)

Not so fluffy as the web gui🙈

Arksine, 2022-05-28:
This appears to be a bug in [Fluidd](https://github.com/fluidd-core/fluidd), you may want to report it there.

matmen, 2022-05-29:
looks like this is mainsail, not fluidd ;)

Arksine, 2022-05-29:
Oops, sorry, you are correct!  I have corrected the post

SchwabJ, 2022-05-29:
Bug is fixed there in beta4. Thank you. My mistake.

closed 2022-06-05 by Arksine