Arksine/moonraker #440: [BUG] Slicer upload

Potattoe, 2022-05-28:
### What happened

Hello, after complete raspberry reinstall i can’t upload files from slicer, i always get this error. I've tried changing ip adress, adding api key, creating new folder for gcodes and changing some settings in moonraker, but nothing helped.
[image]

### Client

Fluidd

### Browser

Firefox

### How to reproduce

I don't know how to reproduce this issue.

Arksine, 2022-05-28:
Most likely you don't have `[octoprint_compat]` in `moonraker.conf`.  The [documentation](https://moonraker.readthedocs.io/en/latest/configuration/#octoprint_compat) has details on its configuration.

Potattoe, 2022-05-28:
I have [octoprint_compat] in moonraker.conf, i tried to reinstall moonraker again and it didn't help, i'm compleatly lost.. :/

Arksine, 2022-05-28:
Please provide `moonraker.log`.  It should provide a clue as to what is going on.

Potattoe, 2022-05-28:
I just reinstalled my whole raspberry again and it's working now, thanks for your help. <3

closed 2022-05-29 by Potattoe