Arksine/moonraker #446: Invalid root path (gcodes)

linghaokeji, 2022-06-05:
### What happened

I tried install klipper+moonraker+fluidd on debian buster and debian bulleyes use kiauh

### Client

Fluidd

### Browser

Chrome

### How to reproduce

install klipper+moonraker+fluidd with kiauh on orangepi's debian system (server version) No error is reported during the installation process

### Additional information

[image]
[image]
[image]

Arksine, 2022-06-05:
Please attach the full `moonraker.log`.  Generally this error occurs when Moonraker has been unable to connect to Klipper.

yu3000, 2022-07-01:
I found the same problem "Invalid root path",the attachment is my log with debug :true
please help, thanks.
[moonraker (1).log](https://github.com/Arksine/moonraker/files/9027776/moonraker.1.log)

Arksine, 2022-07-01:
Moonraker needs to successfully connect to an instance of Klipper in the "ready" state at least once to retrieve the gcode path.  After its received Moonraker will save it in the database.   In the future I plan to change this behavior, where Moonraker dictates the path and Klipper's configuration must match.

FWIW, the log shows that the source has been modified, if the issue persists after fixing Klipper then I'll need you to reproduce with a pristine repo to be able to track down exactly what is happening.

yu3000, 2022-07-01:
thanks for you fast reply.
I have clear my codes and recover to a pristine repo.
This is my current version.

fluidd
v1.18.2
UP TO DATE
klipper
v0.10.0-508-g1636a975
UP TO DATE
moonraker
v0.7.1-597-gd37f91c
UP TO DATE
OS Packages

[moonraker (2).log](https://github.com/Arksine/moonraker/files/9028528/moonraker.2.log)
[klippy (2).log](https://github.com/Arksine/moonraker/files/9028529/klippy.2.log)

yu3000, 2022-07-01:
I want to provide more information.

I can find my mcu id by following commands
<<<<<<<<<<<<<<<<<<<
ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

but I can't connect to my mcu with following message.
<<<<<<<<<<<<<<<<<<
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

yu3000, 2022-07-02:
the problem has been fixed,thanks.

Stars-world, 2022-07-02:
Can you tell me how you solved it? I have the same problem.

Antynik, 2022-07-02:
I have the same problem too.

Arksine, 2022-07-02:
Klipper must be able to enter the ready state before Moonraker can request the gcode path from it.  If you need assistance getting Klipper running the [Klipper Discourse](https://klipper.discourse.group/c/general-discussion/6) is the recommended forum for requesting assistance.  Make sure you include your logs (I suggest both klippy.log and moonraker.log) with your request.

There are also help channels on the Klipper Discord, however the nature of discord can make it more difficult for someone to see it.

Antynik, 2022-07-02:
my mistake was that I configured the board without a printer

yu3000, 2022-07-04:
You should excute the command by the following squence.
<<<<<<<<
sudo service klipper stop
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
sudo service klipper start
>>>>>>>>

Once the mcu connect successfully, the error "Invalid root path" will go.

Arksine, 2022-07-09:
As this issue appears to be resolved I am going to go ahead and close it.  Thanks.

closed 2022-07-09 by Arksine