Arksine/moonraker #447: Regression on the [authorization] parsing?

leandrotoledo, 2022-06-08:
### What happened

After updating Moonraker to the latest version ```v0.7.1-576-g3b76d81``` I am not able to connect to it anymore via Fluidd. When inspecting the https://myprinter.lan/server/info, I get under ```[warnings]```:
```
"Unparsed config section [authorization] detected.  This may be the result of a component that failed to load.  
In the future this will result in a startup error."
```

I have not changed anything on ```moonraker.conf```, especially under the ```[authorization]``` section, it remains:
```
[authorization]
cors_domains:
   https://app.fluidd.xyz
trusted_clients:
   192.168.0.0/16
```

### Client

Fluidd

### Browser

Chrome, Firefox

### How to reproduce

On Fluidd or using ```git pull```, pull the latest Moonraker commits, ```v0.7.1-576-g3b76d81``` which introduced the [LDAP support](https://github.com/Arksine/moonraker/commit/a86cbc77f6182198e1058752a8ee77a516b01b7e#diff-02dfed486292f02d8ff81f38e5b2e9e72e9d8db1129c4f3a35f3b04010f21738)

### Additional information

Any thoughts on implementing a ```stable``` branch for rollout updates?

Thanks for this great software!

leandrotoledo, 2022-06-08:
I should have added that, even as a warning, when trying to connect via Fluidd, I'm getting now the following error: 
```blocked by CORS policy```

I'm using https://app.fluidd.xyz, which is allow-listed in the ```cors_domains``` section, and it was working before the update.

I will try to rollback and report back to confirm if this is indeed a regression.

leandrotoledo, 2022-06-08:
Confirmed that if I:
```
$ git checkout 74d6e3516694c4fbda84ceaffd83a34673fcc87c
$ sudo service moonraker restart
```

I can connect to Moonraker instance again, and there are no warnings issued on /server/info.

Arksine, 2022-06-08:
Having looked into this, there was a bug in the update manager.  I fixed it and reverted the ldap changes.  I will add them back in the future, however users will need to update to the bug fix.

closed 2022-06-17 by Arksine