Arksine/moonraker #450: Maintenance odometer

charlespick, 2022-06-17:
### Is your feature request related to a problem? Please describe

Nevermore carbon needs to be replaced, linear rails need to be lubed, pei sheets and nozzles need to be at least inspected. It would be nice if moonraker could use it's existing history database to calculate print time and days since certain maintenance tasks were performed and send reminders when maintenance is due. 

### Describe the solution you'd like

Alerts when starting a print, notifications in mainsail/klipper, email notifications?, and a general api endpoint to display that data in clients

### Describe alternatives you've considered

notes apps lol. it's not great

Martyfree123, 2022-06-24:
+1 for this. Would be amazing.

karlfife, 2024-03-10:
This is a great idea.   It could lincorporate something like a "Trip Odometer" for filament usage, so you measure how much you've burn in a given project, or how much wear you have on a given nozzle etc.