Arksine/moonraker #454: non stop disconnects

Fuckingnameless, 2022-07-04:
### What happened

since updating a couple days ago it disconnects frenetically, i will attach the corresponding part of the log below

### Client

Mainsail

### Browser

Brave

### How to reproduce

do nothing

### Additional information

2022-07-04 14:23:04,224 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /access/login
2022-07-04 14:23:04,225 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /access/logout
2022-07-04 14:23:04,227 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /access/refresh_jwt
2022-07-04 14:23:04,229 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET POST DELETE) /access/user
2022-07-04 14:23:04,237 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /access/users/list
2022-07-04 14:23:04,238 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /access/user/password
2022-07-04 14:23:04,240 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET POST) /access/api_key
2022-07-04 14:23:04,247 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /access/oneshot_token
2022-07-04 14:23:04,249 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /access/info
2022-07-04 14:23:04,250 [moonraker.py:load_component()] - Component (authorization) loaded
2022-07-04 14:23:04,259 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /api/version
2022-07-04 14:23:04,260 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /api/server
2022-07-04 14:23:04,263 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /api/login
2022-07-04 14:23:04,267 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /api/currentuser
2022-07-04 14:23:04,268 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /api/settings
2022-07-04 14:23:04,272 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /api/job
2022-07-04 14:23:04,279 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /api/printer
2022-07-04 14:23:04,280 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /api/printer/command
2022-07-04 14:23:04,282 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /api/printerprofiles
2022-07-04 14:23:04,284 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /api/files/moonraker/(?P<relative_path>.+)
2022-07-04 14:23:04,294 [moonraker.py:load_component()] - Component (octoprint_compat) loaded
2022-07-04 14:23:04,300 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET DELETE) /server/history/job
2022-07-04 14:23:04,301 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.history.get_job, server.history.delete_job
2022-07-04 14:23:04,301 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /server/history/list
2022-07-04 14:23:04,302 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.history.list
2022-07-04 14:23:04,303 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /server/history/totals
2022-07-04 14:23:04,304 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.history.totals
2022-07-04 14:23:04,304 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /server/history/reset_totals
2022-07-04 14:23:04,316 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.history.reset_totals
2022-07-04 14:23:04,326 [moonraker.py:load_component()] - Component (history) loaded
2022-07-04 14:23:04,445 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/update/moonraker
2022-07-04 14:23:04,447 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.update.moonraker
2022-07-04 14:23:04,447 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/update/klipper
2022-07-04 14:23:04,448 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.update.klipper
2022-07-04 14:23:04,449 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/update/system
2022-07-04 14:23:04,450 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.update.system
2022-07-04 14:23:04,450 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/update/client
2022-07-04 14:23:04,456 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.update.client
2022-07-04 14:23:04,457 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/update/full
2022-07-04 14:23:04,458 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.update.full
2022-07-04 14:23:04,458 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /machine/update/status
2022-07-04 14:23:04,459 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.update.status
2022-07-04 14:23:04,459 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/update/recover
2022-07-04 14:23:04,460 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.update.recover
2022-07-04 14:23:04,461 [moonraker.py:load_component()] - Component (update_manager) loaded
2022-07-04 14:23:04,492 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/timelapse/render
2022-07-04 14:23:04,494 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.timelapse.render
2022-07-04 14:23:04,494 [app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /machine/timelapse/saveframes
2022-07-04 14:23:04,506 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.timelapse.saveframes
2022-07-04 14:23:04,507 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET POST) /machine/timelapse/settings
2022-07-04 14:23:04,509 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: machine.timelapse.get_settings, machine.timelapse.post_settings
2022-07-04 14:23:04,509 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /machine/timelapse/lastframeinfo
[... 134 lines cut ...]
2022-07-04 14:25:42,662 [websockets.py:open()] - Websocket Opened: ID: 1749746992, Proxied: False, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.5060.66 Safari/537.36, Host Name: 192.168.15.3
2022-07-04 14:25:42,678 [websockets.py:_handle_identify()] - Websocket 1749746992 Client Identified - Name: mainsail, Version: 2.2.1, Type: web
2022-07-04 14:25:46,156 [websockets.py:on_close()] - Websocket Closed: ID: 1749810928 Close Code: None, Close Reason: None, Pong Time Elapsed: 4.58
2022-07-04 14:25:46,183 [websockets.py:_process_messages()] - Websocket closed while writing: 1749810928
2022-07-04 14:25:57,352 [websockets.py:on_close()] - Websocket Closed: ID: 1749746992 Close Code: None, Close Reason: None, Pong Time Elapsed: 14.69
2022-07-04 14:25:57,364 [app.py:log_request()] - 101 GET /websocket (192.168.15.2) [_TRUSTED_USER_] 6.86ms
2022-07-04 14:25:57,365 [websockets.py:open()] - Websocket Opened: ID: 1749788432, Proxied: False, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.5060.66 Safari/537.36, Host Name: 192.168.15.3
2022-07-04 14:25:57,377 [websockets.py:_handle_identify()] - Websocket 1749788432 Client Identified - Name: mainsail, Version: 2.2.1, Type: web
2022-07-04 14:26:11,602 [klippy_connection.py:wait()] - Request 'gcode/script' pending: 60.00 seconds
2022-07-04 14:26:12,690 [websockets.py:on_close()] - Websocket Closed: ID: 1749999824 Close Code: None, Close Reason: None, Pong Time Elapsed: 4.33
2022-07-04 14:26:12,701 [app.py:log_request()] - 101 GET /websocket (192.168.15.2) [_TRUSTED_USER_] 3.89ms
2022-07-04 14:26:12,702 [websockets.py:open()] - Websocket Opened: ID: 1749761264, Proxied: False, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.5060.66 Safari/537.36, Host Name: 192.168.15.3
2022-07-04 14:26:13,053 [file_manager.py:get_file_list()] - Updating File List <config>...
[... cut ...]
2022-07-04 14:26:19,660 [websockets.py:on_close()] - Websocket Closed: ID: 1749761264 Close Code: None, Close Reason: None, Pong Time Elapsed: 6.96
2022-07-04 14:26:19,673 [app.py:log_request()] - 101 GET /websocket (192.168.15.2) [_TRUSTED_USER_] 5.64ms
2022-07-04 14:26:19,674 [websockets.py:open()] - Websocket Opened: ID: 1750121968, Proxied: False, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.5060.66 Safari/537.36, Host Name: 192.168.15.3
2022-07-04 14:26:19,783 [file_manager.py:get_file_list()] - Updating File List <config>...

Arksine, 2022-07-04:
It is necessary to attach the full log for me to make a complete diagnosis, however based on the snippet it appears to be a local networking error.   Neither Moonraker nor Mainsail requesting the websocket disconnection, its just dropping.

Fuckingnameless, 2022-07-08:
> It is necessary to attach the full log for me to make a complete diagnosis, however based on the snippet it appears to be a local networking error. Neither Moonraker nor Mainsail requesting the websocket disconnection, its just dropping.

eee was the culprit it seems

https://forums.raspberrypi.com/viewtopic.php?t=305820

Arksine, 2022-07-08:
Thanks for the update.  If the issue is EEE on a Pi 3B+ then that suggests a regression may have  been introduced in the kernel driver.  I'll keep this info tucked away, I may look into an faq for troubleshooting connection issues.

closed 2022-07-09 by Arksine