Arksine/moonraker #457: directive defined in included conf file cannot be overriden in main moonraker.conf

kmarty, 2022-07-07:
### What happened

Hi,
when any(?) directive is defined in conf file used in "[include ...]", it cannot be overriden by the same section and directive in main _moonraker.conf_

It looks like the "_If a section is duplicated in an included file the options from both sections will be merged, with the latest file parsed taking precedence_" works only between included files, not with definitions in "main" _moonraker.conf_, used in `-c` argument.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

E.g.:
In ./config/some-ratos-default.conf is defined:
```
[file_manager]
enable_object_processing: False
```

and when in "main" _moonraker.conf_ is:
```
[include config/some-ratos-default.conf]

[file_manager]
enable_object_processing: True
```

the result is `enable_object_processing = False`

But when I, instead of redefining this directive in "main" _moonraker.conf_, create another conf file, e.g. _my-changes.conf_, with:
```
[file_manager]
enable_object_processing: True
```
and in "main" _moonraker.conf_ is:
```
[include config/some-ratos-default.conf]

[include my-changes.conf]
```
the result is `enable_object_processing = True`

### Additional information

I tried e.g. `enable_debug_logging` in `[server]` section and result is the same, so it is general thing, not `[file_manager]` or `enable_object_processing` specific.

Arksine, 2022-07-07:
This is expected behavior, `moonraker.conf` is the entry point, in other words it is the first configuration file parsed.  Thus options defined in any included file will override options defined in `moonraker.conf`.

kmarty, 2022-07-07:
Oh.
Kinda unexpected behaviour (definitely different than e.g. in Klipper’s printer.cfg), but ok.

miklschmidt, 2022-07-07:
> This is expected behavior, `moonraker.conf` is the entry point, in other words it is the first configuration file parsed. Thus options defined in any included file will override options defined in `moonraker.conf`.
> 
> Keep in mind that included files are not parsed until parsing is complete in the existing file. So putting an `include` directive before `file_manager` does not mean that options in `file_manger` take precedence. A full configuration file is parsed, then its included files are parsed.

Any specific reason why the behavior isn't identical to klipper? Is it necessary to have to teach two different paradigms for syntax that look identical and are very much associated with one another?

miklschmidt, 2022-07-07:
> I used the config merging functionality native to Python's configparser when implementing it. Klipper has a robust wrapper around the configparser which allows it do things like reorder how the config is merged and implement the SAVE_CONFIG section.

Right, too much work to reuse it?
 
> I was really hesitant to add include directives to be honest. Its going to make life more difficult when we pursue a UI driven configuration.

I'm working on that, includes makes it easier, especially when it comes to customizing macro behavior, not harder. But slightly different use case, as klipper works differently. Not far enough to have it bite my ass yet at least.

Will have to rethink the moonraker include in RatOS, it's causing a bit of confusion when their "overrides" don't do anything. I'd have to use an additional include for them to make their changes in or something like that.

Arksine, 2022-07-07:
> Right, too much work to reuse it?

Yeah, would be a significant amount effort to port it over and include the Moonraker specific functionality that currently exists..  That said, I'm sure its possible to modify Moonraker's include functionality to buffer the config files similar to how Klipper does it.  I'm willing to look into it.

miklschmidt, 2022-07-08:
> That said, I'm sure its possible to modify Moonraker's include functionality to buffer the config files similar to how Klipper does it. I'm willing to look into it.

That would be awesome for sure!

Arksine, 2022-07-27:
I have made this change.  Everything tests well on my end, however given that this affects user configuration it would be a good idea others to pull it and make sure it behaves.

kmarty, 2022-08-03:
Hi, I tried that my exact issue I've had and it works for me now.

miklschmidt, 2022-08-05:
Works as expected :+1: very nice! @kmarty i guess you can close this one now :)

closed 2022-08-05 by kmarty