Arksine/moonraker #459: No power buttons on mainsail and fluid

xmannx, 2022-07-09:
### What happened

I have new instalation mainsail on ma raspberry pi 3b+
I added to moonraker.cfg this line:

[power printer]
type: gpio
pin: gpio6 # uses pin 6 on gpiochip0
off_when_shutdown: True
restart_klipper_when_powered: true
initial_state: off

[power printer_led]
type: gpio
pin: !gpiochip0/gpio12 # inverts pin
initial_state: off

In menu missing power buttons to power on printer and printer_led

When i instaled fluidd, in menu was power button, but was empty. Not line "printer" and line "printer_led"

here is 3 lines errors from moonraker.log:

2022-07-09 20:43:49,714 [moonraker.py:add_warning()] - Failed to load power device [power printer]
Error parsing option (pin) from section [power printer]
2022-07-09 20:43:49,715 [moonraker.py:add_warning()] - Failed to load power device [power printer_led]
Error parsing option (pin) from section [power printer_led]

### Client

Mainsail

### Browser

Chrome

### How to reproduce

missing power buttons

Arksine, 2022-07-09:
Moonraker does not support inline comments in its configuration.  Remove them and it should work.   If not, it will be necessary to attach the full moonraker.log to diagnose the issue.

xmannx, 2022-07-09:
it work. Inline comments was problem. I copy it from exsamples on web sites https://moonraker.readthedocs.io/en/latest/configuration/

thank you

closed 2022-07-09 by Arksine