Arksine/moonraker #471: Component 'webcam' failed to load with error: Error running shell command: '['iwgetid']'

adman953, 2022-07-24:
### What happened

I updated a hard wired non wifi rapsberry with fluidd and now i get the error message:

Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
webcam
Moonraker warnings found.
Component 'webcam' failed to load with error: Error running shell command: '['iwgetid']'
Moonraker plugin configuration can be found [here.](https://docs.fluidd.xyz/configuration/moonraker)

I have tried updating the configuration to include the webcam section and tried a number of different settings but nothing works. I have checked the log and have found that iwgetid is wifi dependent and since i do not use wifi it does not work. the webcam does display in fluidd but I have this giant orange error across the top of the interface hat I do not like.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

best guess is update an hard wired raspberry with a webcam and see if the problem reproduces
[moonraker (1).log](https://github.com/Arksine/moonraker/files/9176488/moonraker.1.log)

marco-117, 2022-07-24:
I had the same issue i reinstalled moonraker like descripted in the issue #349 with `~/moonraker/scripts/install-moonraker.sh`

After that i removed unsued packages with `sudo apt autoremove` like in the installation process metioned.

Now everything works fine, with no errors or warnings.

Arksine, 2022-07-24:
That is unexpected.   It suggests that the `iwgetid` command isn't located in the user's path.  This isn't a critical bug, things should still work ok, however annoying nonetheless.  I'll see if I can come up with a workaround.

Arksine, 2022-07-24:
I pushed a commit that should handle the exception if `iwgetid` fails.  This should allow the webcam module to load correctly

adman953, 2022-07-24:
that fixed it, thank you so much for the quick response

Arksine, 2022-07-27:
Thanks, I'll close this since it is resolved

closed 2022-07-27 by Arksine