Arksine/moonraker #473: Changes to include behavior

Arksine, 2022-07-27:
The behavior of Moonraker's include directives has changed.  Include directives are now parsed in place as if the included file were copied into its parent.  This is how Klipper and other applications handle include directives.

To override configuration in `moonraker.conf` the include directive must be placed *after* the section is defined in `moonraker.conf`.  Generally speaking this would be at the bottom of the file.  If the objective is for sections in `moonraker.conf` to take precedence then the include directive should be placed at the top of the file.

Moonraker's configuration now supports inline comments as well.

dxgldotorg, 2022-08-28:
So just put includes where you want to insert the contents of the included file, just like a header in C/C++?

dcp3D, 2022-09-27:
could anyone please provide an example that I might be able to copy into my printer.cfg file I am unsure how to do this?

dxgldotorg, 2022-09-28:
> could anyone please provide an example that I might be able to copy into my printer.cfg file I am unsure how to do this?

Isn't printer.cfg for the Klipper settings, which are outside the scope of Moonraker?

closed 2022-10-14 by Arksine