Arksine/moonraker #481: [FR] Pipe through Klipper Estimator

Sineos, 2022-08-09:
### Is your feature request related to a problem? Please describe

Currently the print time guesses are just that - guesses based on slicer information and (I assume) layer times of the previous layers.

### Describe the solution you'd like

Pipe the gcode transparently through [Klipper Estimator](https://github.com/Annex-Engineering/klipper_estimator) to return accurate print time forecasts to the clients

### Describe alternatives you've considered

Leave it as is. Anyway takes as long as it takes 😉

gauravmm, 2022-08-10:
I'm happy to do this as a moonraker plugin -- it will fit nicely with the  gcode server plugin I was writing.

Dreistein75, 2022-11-18:
Is there a chance this can be implemented like the EXCLUDE_OBJECT post-processing? That would be very neat

gauravmm, 2022-11-19:
The current idea is that this is dynamically calculated (and cached) by a moonraker plugin, so no slicers need to be involved, it can pull the latest Klipper configuration information automatically, and it can recompute the estimate when the Klipper configuration changes.

lbibass, 2023-01-30:
Bump! Any updates on this?

gauravmm, 2023-01-31:
I'm almost done writing my PhD dissertation, I'll start work on this in mid-Feb. Sorry about the delay!

lbibass, 2023-02-07:
> I'm almost done writing my PhD dissertation, I'll ~start~ resume work on this in mid-Feb. Sorry about the delay!

No worries! Congrats on nearly being finished with your PhD! Good luck!

richfelker, 2023-11-27:
Ping. Any progress?

clinta, 2024-01-21:
Using klipper estimater as a post processing script in my slicer results in very accurate estimates if I look at the slicer time remaining in mainsail. Makes me wish I could just disable the built in time remaining, cause it's basically pointless.

nelsongraca, 2024-03-08:
I have written a plugin to run gcode on upload through klipper estimator, still a little rough but it's working.

@Arksine would you be willing to add this in or would you prefer for it to be a plugin?

CharlemagneLasse, 2024-09-30:
@Arksine this would improve my life so much. Just happens to often that I go to the printer (on the incorrectly estimated time by the slicer) just to figure out that I could have stayed an hour longer at home instead of waiting for the 3D printer in the workshop to finally finish. I don't have any control about the slicer because I am often not the person who sliced and started the print.

@nelsongraca I am not sure how this is handled here but maybe having some RFC/Draft to discuss this further would help

nelsongraca, 2024-09-30:
@CharlemagneLasse I have it working on my printer, I can publish the code somewhere or submit a PR.

nelsongraca, 2024-10-22:
to those who might be interested, opened PR, still a draft, also be aware that my python skills are not the best

nelsongraca, 2024-10-26:
it's up for review, CI passes, so if anyone wants to have a look please do