Arksine/moonraker #483: Power: on_when_job_queued not working as described

btastic, 2022-08-10:
### What happened

Hello,

I just finished setting up my tasmota device with moonraker, to turn on my printer via moonraker. However, the config states that it can work like this:

```
on_when_job_queued: False
#   If set to True the device will power on if a job is queued while the
#   device is off.  This allows for an automated "upload, power on, and
#   print" approach directly from the slicer, see the configuration example
#   below for details. The default is False.
```

I was eager about the "upload, power on, and print" part. However, when using PrusaSlicer, the print does not get **queued** when using the "Upload and print" feature. It get's **uploaded**, but not **queued**.
When **queuing** the job via the web interface, the printer does indeed turn on.

Not sure if this is a bug or whether PrusaSlicer is not the right choice. But for now it is not working as described.

### Client

Mainsail

### Browser

Microsoft Edge

### How to reproduce

- Set up a power plug
- Set on_when_job_queued to True
- Slice something in PrusaSlicer
- Use Upload and Print from the Slicer
- Job does get **uploaded**
- Job does not get **queued**

### Additional information

```ini
[power Printer]
type: tasmota
address: 192.168.0.23
off_when_shutdown: False
on_when_job_queued: True
locked_while_printing: True
restart_klipper_when_powered: True
restart_delay: 1
```

Arksine, 2022-08-10:
I suspect that there is missing configuration.  Make sure you followed the [full instructions](https://moonraker.readthedocs.io/en/latest/configuration/#power-on-g-code-uploads) required to enable this feature.

btastic, 2022-08-10:
Thanks. This will be it. I went straight to the power configuration description without checking much else. Sorry.

Thanks again!

closed 2022-08-10 by btastic

Primer-Merc, 2022-12-13:
Typo here guys...

[image]