Arksine/moonraker #487: The exclude object post processing script seems does not recognise the format labeld by Cura5.1.0

SHKinsem, 2022-08-18:
### What happened

As the issue I descibed [here](https://github.com/mainsail-crew/mainsail/issues/1032).
I uploaded the gcode file sliced with Cura to the web and checked the contents. I found no extra lines were added by the script while the file sliced with prusa did.

### Client

Mainsail

### Browser

Microsoft Edge

### How to reproduce

Using the file sliced with Cura5.1

### Additional information

I find out that there is a few differences about the labels labeld by both slicer.
Prusa labeld the object both at start print and end print, 
e.g.

>  ; printing object xyzCalibration_cube.stl id:0 copy 0
> ....GCODES....
> ; stop printing object xyzCalibration_cube.stl id:0 copy 0
>  ; printing object xyzCalibration_cube.stl id:1 copy 0
> ....GCODES....
> ; stop printing object xyzCalibration_cube.stl id:1 copy 0

But Cura labels the start print of the object only,
e.g.

> ;MESH:xyzCalibration_cube.stl
> ....GCODES...
> ;MESH:xyzCalibration_cube.stl(1)
> ....GCODES....

Arksine, 2022-08-21:
You'll need to reproduce this issue and attach moonraker.log.  With that I should be able to determine if the issue is in Moonraker's metadata processor or if its in the `preprocess-cancellation` dependency.

SHKinsem, 2022-08-22:
Sorry, I tried to print something again today and it seems worked.....
I am closing the issue.

closed 2022-08-22 by SHKinsem

RagingRoosevelt, 2023-01-16:
Not OP, but I'm having issues with Cura 5.3.0, klipper v0.11.0-66-g4671cf2d, mainsail v2.4.1, moonraker v0.7.1-806-ga154c5f.  I've read through the GCODE file and see these lines:

```
;MESH:<stl name here>
...
;TYPE:<type here>
...
```
throughout the file

When I look over my moonraker logs, I see
```
2023-01-15 23:23:45,021 [shell_command.py:run_with_response()] - /home/pi/moonraker-env/bin/python: Object Processing is disabled
/home/pi/moonraker/moonraker/components/file_manager/metadata.py:558: DeprecationWarning: ANTIALIAS is deprecated and will be removed in Pillow 10 (2023-07-01). Use Resampling.LANCZOS instead.
  im.thumbnail((32, 32), Image.ANTIALIAS)

2023-01-15 23:23:45,021 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/file_manager/metadata.py -p /home/pi/gcode_files -f "(VT-06)_Sentry-black_parts.gcode" -u "/tmp/moonraker.upload-1563967.mru") successfully finished
```

but in `moonraker.conf`, I have
[image]

and in `printer.cfg` I have
[image]

RagingRoosevelt, 2023-01-16:
NVMD, Turns out you have to reboot the whole Pi between turning on including these settings and uploading the file before Moonraker will process for you.