Arksine/moonraker #499: Cant update Moonraker - Git issue

PhilBaz, 2022-09-09:
### What happened

I can update from Mainsail.... see image. Is this on my side?

[image]

### Client

Mainsail

### Browser

Chrome

### How to reproduce

press update button

Arksine, 2022-09-10:
You have a corrupt git repo.  If Mainsail presents a recovery option you could try it (perform a "hard" recovery if available).  Otherwise can use ssh to do a [manual recovery](https://moonraker.readthedocs.io/en/latest/installation/#recovering-a-broken-repo).

Repo corruption typically occurs when power is removed from the host machine (Pi, SBC, etc) without properly shutting down.

PhilBaz, 2022-09-10:
@Arksine 

Thank you. Manual recovery worked. I didn't see an option in MS. The recovery process also prompted me to update pip, and I did.

WARNING: You are using pip version 21.3.1; however, version 22.2.2 is available.
You should consider upgrading via the '/home/pi/moonraker-env/bin/python -m pip                install --upgrade pip' command.

closed 2022-09-10 by PhilBaz