Arksine/moonraker #502: restart_klipper_when_powered does not work

andreasmwestberg, 2022-09-15:
### What happened

I have a MQTT controlled power switch for my 3d printer and I've configured Moonraker so I can turn it on and off in fluidd web gui.

I've followed [power-on-g-code-uploads](https://moonraker.readthedocs.io/en/latest/configuration/#power-on-g-code-uploads), when I press 'Upload and print' in SuperSlicer the switch turns on and the printer gets power but the Klipper service doesn't connect to the MCU even though I'm using 'restart_klipper_when_powered: True'.

If I click 'FIRMWARE RESTART' in Fluidd, it connects and starts printing the gcode in the queue.

I've tried different restart_delay and commenting out everything but essentials in config.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

- Have a MQTT controlled power switch
- Use my Moonraker config
- Do 'Upload and print' in SuperSlicer

### Additional information

console
```
11:11:40 !! Lost communication with MCU 'mcu'
11:11:40 // Lost communication with MCU 'mcu'
         // Once the underlying issue is corrected, use the
         // "FIRMWARE_RESTART" command to reset the firmware, reload the
         // config, and restart the host software.
         // Printer is shutdown
```
moonraker.log
```
--------------------Moonraker Log Start--------------------
config_file: /home/pi/klipper_config/moonraker.conf
startup_warnings: []
log_file: /home/pi/klipper_logs/moonraker.log
software_version: v0.7.1-660-g5d856b9
python_version: 3.9.2 (default, Mar 12 2021, 04:06:34)  [GCC 10.2.1 20210110]

#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
enable_debug_logging = False
max_upload_size = 1024

[mqtt]
address = ****
port = 1883
username = ****
password = ****
status_objects = 
	webhooks
	toolhead
	extruder
	heater_bed
	fan

[file_manager]
log_path = ~/klipper_logs
config_path = ~/klipper_config
queue_gcode_uploads = True

[job_queue]
[... 297 lines cut ...]
control: pid
pid_Kp=75.097
pid_Ki=0.770
pid_Kd=900.000

[fan]
pin: PA0

[mcu]
serial: /dev/ttyAMA0
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[board_pins]
aliases:
  EXP1_1=PC6,EXP1_3=PB10,EXP1_5=PB14,EXP1_7=PB12,EXP1_9=<GND>,
  EXP1_2=PB2,EXP1_4=PB11,EXP1_6=PB13,EXP1_8=PB15,EXP1_10=<5V>,
  PROBE_IN=PB0,PROBE_OUT=PB1,FIL_RUNOUT=PC6
```

Creo2005, 2022-09-15:
Possible without any MQTT:
in my moonraker.conf:
[power XXX]
type: homeassistant
address: 192.168.1.193
port: 8123
token: xxxxxxxxxxxxxxxxxx
protocol: http
device: switch.printer_rozetka
domain: switch
off_when_shutdown: True
off_when_shutdown_delay: 600
on_when_job_queued: True
restart_klipper_when_powered: True
restart_delay: 2
locked_while_printing: True

andreasmwestberg, 2022-09-15:
@Creo2005 I've just tested switching to homeassistant instead of MQTT, and it works

so then the problem is that restart_klipper_when_powered does not work when using MQTT

Arksine, 2022-09-18:
Please attach the logfiles themselves rather than copy and paste their contents.  I can tell parts of the configuration in `moonraker.log` were modified, in particular it appears that `restart_klipper_when_powered` was added to the log after the fact.  If you need to hide credentials you can use the secrets module, and its acceptable to hide or remove the host name in the mqtt section, but please do not modify or add anything else to the log. 

I'll see if I can reproduce the mqtt issue locally, from the snippet about it appears the the device is receiving an external request to power on/off.  If something else is toggling the device it would bypass Moonraker's ability to perform actions based on the request (such as restarting Klipper).

Arksine, 2022-11-11:
I was unable to reproduce this and no additional logs have been provided.  If this someone can reproduce this with log files feel free to attach and reopen the issue.  Thanks.

closed 2022-11-11 by Arksine