Arksine/moonraker #510: Moonracker wont connect with MCU until full reboot and uplug/plug USB

grabbershelby, 2022-09-20:
### What happened

I have a Ender 3 S1 Pro connected to a Raspberry Pi 4B Running Klipper.  I have my moonraker.conf setup to toggle my printer power using a tasmota plug.  When my printer is powered off, the device is not visible to the host when it is turned back on.  I have referend issue #835 and various other posts online to no avail.  This seems different.  Not so much a auto reconnect issue but something else.

Before toggling power off:
pi@mainsail:~ $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
pi@mainsail:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

When power is turned back on I get an error that cannot connect to MCU, when I go to putty and check there is not directory  /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0.

This is reflected in the moonracker and klippy log.  The only way I am able to regain connection is the reboot everything, and I must unplug the printer USB and plug it back in.

I created the rule below but this did not help.
## rule to restart klipper when the printer is connected via usb
SUBSYSTEM=="usb", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", ACTION=="add", RUN+="/bin/sh -c '/bin/echo RESTART > /tmp/printer'"

I will attach my klippy and moonracker logs as well as my configs.  Any help would be very much apprecited.

[power Ender-3 S1 Pro]
type: tasmota
on_when_job_queued: True
locked_while_printing: True
restart_delay: 5
address: xxxxxxxxxxxxxxx

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

[moonraker.log](https://github.com/Arksine/moonraker/files/9603179/moonraker.log)
[klippy.log](https://github.com/Arksine/moonraker/files/9603180/klippy.log)

### Client

Mainsail

### Browser

Microsoft Edge

### How to reproduce

Toggle power to device via mainsail/moonracker.

Arksine, 2022-09-21:
This sounds like a Kernel issue.  Unfortunately there is nothing I can do within Moonraker to facilitate a fix here.  You may have more luck asking about your specific device in the Klipper discord, its possible someone else with your printer has encountered a similar problem and has a fix.

grabbershelby, 2022-09-21:
I’m thinking the issue is on my hardware side. Last night I rolled my printer back to marlin and tried to connect to octoprint on my pi and could not establish a connection. Then when I connected my printer to my laptop, windows gave me an error as well. It stated the device had malfunctioned and was unrecognized. When I installed klipper, in an effort to clean up the cable management, I soldered a connection onto the mainboard usb header so that I could keep my wiring internal to my printer’s enclosure.  I’m wondering if i have a failed solder job that is causing the connection issue. Just for reference, I have attached a picture of where i connected my USB cable. Hopefully this did not damage the mainboard and I can fix or just remove my connection.

[image]

grabbershelby, 2022-09-21:
I just checked my mainboard and the two data pins on the USB headed where in fact shorted. Must have been just enough solder there to cause the issue. It was not even visible. Cleaned it up and got rid of the short. I do not think this would have damaged any hardware, just caused some serious connection issues. I’m surprised it worked at all like this. I didn’t have time to troubleshoot any further. I now no longer get any error messages when connected to my laptop. I will see if I can establish communication with with my pi and get up and running with klipper again this evening.

grabbershelby, 2022-09-24:
Confirmed that my issue was due to my USB connection to the USB header. I re-soldered my PCB pins and everything is running as expected without any connection issues.

closed 2022-09-24 by grabbershelby