Arksine/moonraker #516: Folder/Path configuration change

Arksine, 2022-10-03:
An upcoming addition to Moonraker will change how files and folders are configured and managed in Moonraker.  All folders and files managed by Moonraker will be contained within a `data path` configured in the command line.  Options such as `config_path` and `log_path`, `secrets_path`, etc. will be deprecated.

A detailed description of the changes can be found in pull request #491.  The current plan is to merge #491 on Friday, October 7th.  The changes to the configuration file and system service are automated for most installations, however the user's root password may be necessary if Moonraker does not have sudo access.   In this scenario Moonraker will generate a `Sudo Password Required` announcement:

[image]

[image]

The incomplete update will generate several warnings so the announcement may not be at the very top of the notification panel.  The user should click through the link (the title in Mainsail, `More Information` in Fluidd), which will take you to Moonraker's landing page:

[image]

Here Moonraker will request the password for the detected user (in the example above it is `pi`).  After submission Moonraker will finish the update and prompt the user to close the tab/window:

[image]

A Moonraker instance running on a non-debian distro or otherwise customized installation will likely need to perform these changes manually.  It is possible to disable the automated changes in the `machine` configuration:
```ini
[machine]
validate_service: False
validate_config: False
```

chrisbdaemon, 2022-10-03:
Can the sudo command be run manually or do we have to do it through moonraker?

Arksine, 2022-10-04:
@chrisbdaemon It will be possible to manually perform the update via ssh after the PR is merged if that is your preferred method:
```
cd ~/moonraker
sudo service moonraker stop
git pull
./scripts/install-moonraker.sh -f
```

Alternatively, its possible to temporarily set the sudo password in the `[machine]` section of `moonraker.conf` prior to the update.  I would recommend using a [secrets](https://moonraker.readthedocs.io/en/latest/configuration/#secrets) file if you do this. For example, create a file `~/moonraker.secrets` with the following contents:

```ini
# moonraker.secrets

[sudo]
# replace "mypass" with  your password
password: mypass
```

Then in `moonraker.conf`:

```ini
# moonraker.conf
[secrets]
secrets_path: ~/moonraker.secrets

[machine]
sudo_password: {secrets.sudo.password}
```
After the update is complete you can remove the `sudo_password` option from the `machine` section.  You will notice that the `secrets_path` was automatically removed and a symbolic link has been created to that file in `~/printer_data`.   If you desire you can delete that symbolic link and the original secrets file.  Otherwise you can just remove the `[sudo]` section from the secrets file.

MarcoBugo, 2022-10-05:
whenever I click the mainsail title i reach this git and not the moonracker root request why?

Arksine, 2022-10-05:
The changes haven't been merged yet, you are clicking through the announcement generated by this issue.  The currently plan is to merge on October 14th.  At some point after that you will update, which will send the `Sudo Password Required` notification if necessary.

Arksine, 2022-10-06:
FWIW, after testing I found an issue when attempting to use the install script to upgrade...it can't bring over the old database, so the process does not complete successfully.  To provide an option for users that wish to finish the upgrade via ssh I have added a script, `finish-upgrade.sh`.  This script will first check to make sure that Moonraker is awaiting the sudo password.  If that condition is met, it will prompt the user for the password and use it to finish the upgrade.

Eclypse3D, 2022-10-14:
Well this broke my system, it boots up, and now I have no config files, when I try to print I get file not found and the only messages I have in my notifications are the update ones, this was after entering sudo password.

EricZimmerman, 2022-10-14:
Working on this now too. There was 7 notifications, totally gone after a restart, so I can't see what they were

Eclypse3D, 2022-10-14:
I can see what they are they just point me to these pages, but none of these say anything about what to do after the system changes, just cryptic lines of change information.  I do not even know how the printer is able to boot into klipper, I cannot view my configs unless I ssh to them, and I do not know what is changed enough to know how to point paths that are missing or changed lol....

EricZimmerman, 2022-10-14:
once i added sudo password, things just worked and no more warnings, etc. seems like it cleaned up a lot of the stuff

i am using mailsail with klipper on an otherwise default install for a voron 2.4.

Eclypse3D, 2022-10-14:
Yeah I put in the sudo password it closed that screen and never did anything else just left me looking at the welcome to moonraker page.  Now nothing.  Printer boots up, I can control it via dashboard but cannot do anything with it other than change temps, look at updates that say "uptodate" etc.  but cannot print, view logs, etc.  I am not sure if I am supposed to move a folder, etc.  Was just going to finish printing my Voron tonight, looks like thats a bust.

EricZimmerman, 2022-10-14:
you can check the systemd config to see where the env file is and update that for your setup. i verified mine and it looked good. things look like they did yesterday for me now, but i added the sudo password thing manually to the config vs the gui popup

Arksine, 2022-10-14:
> Yeah I put in the sudo password it closed that screen

Try to reload the landing page.  It sounds like you didn't click the submit button and instead clicked out of the dialog box.  This will close the dialog.  When the update completes successfully it will present a popup letting you know.

Eclypse3D, 2022-10-14:
How do I reload that?

Arksine, 2022-10-14:
You can check your notifications to see if it is still in there.  Otherwise you can go to `http://mainsailos.home:7125`.  Replace mainsailos.home with the correct domain

Eclypse3D, 2022-10-14:
all I have is the main page it is not asking for any sudo password now, it just shows 
[image]

JDWarner, 2022-10-14:
I don't have any page at all, Moonraker reports a bad gateway and when trying to fix via SSH I get
[image]

EricZimmerman, 2022-10-14:
mine does not have any warnings, so i guess it fixed itself

[image]

@JDWarner i got the same. i didnt bother with that. i did do the sudo fix and reboot one.

Eclypse3D, 2022-10-14:
I did not skip adding the sudo password, I hit submit and it brought me to the same page from above, there are some errors on that page stating 

Failed to link subfolder 'database' to '/home/pi/.moonraker_database'. Folder 'database' already exists. This conflict requires manual resolution.

GCode path received from Klipper does not match expected path: /home/pi/printer_data/gcodes

Arksine, 2022-10-14:
@Eclypse3D 

Try to ssh in to the Pi and run the following:
```
rm -rf ~/printer_data
sudo service moonraker restart
```

JDWarner, 2022-10-14:
My Moonraker continues to refuse to connect on [correct_ip_address] port 7125.  Bad gateway.

Eclypse3D, 2022-10-14:
This is what I got...

sudo service moonraker restart
sh: 0: getcwd() failed: No such file or directory
Job for moonraker.service failed because of unavailable resources or another system error.
See "systemctl status moonraker.service" and "journalctl -xe" for details.

JDWarner, 2022-10-14:
I think the actual failure is a little above that, here's my log - I think Moonraker is no longer finding its necessary files after the update.
[image]

Eclypse3D, 2022-10-14:
Now Klipper cannot connect to moonraker at all

Arksine, 2022-10-14:
@Eclypse3D Ok, we are going to have to do a manual install to fix it.  The following assumes:
- Your config path is `~/klipper_config/moonraker.conf`
- Your log path is `~/klipper_logs/moonraker.log

```
~/moonraker/scripts/install-moonraker.sh -c ~/klipper_config/moonraker.conf -l ~/klipper_logs/moonraker.log
```

After this is done you will need to apply a fix for the database
```
sudo service moonraker stop
cd ~/printer_data
rm -rf database
ln -s ~/.moonraker_database database
sudo service moonraker start
```

Arksine, 2022-10-14:
@JDWarner  Try the instructions I just gave above, with a small change to the install script:
```
rm -rf ~/printer_data
~/moonraker/scripts/install-moonraker.sh -c ~/klipper_config/moonraker.conf -l ~/klipper_logs/moonraker.log -f
```

JDWarner, 2022-10-14:
Okay, what actually fixed it (just before your message) was deleting printer_data, a hard recovery which restored `printer_data`, and then hardlinking `klipper_config/moonraker.conf` to where the service failure error was expecting it.

Arksine, 2022-10-14:
Interesting.  The error appeared to be because it was unable to find the environment file for the service.  Now that you have access, could you donwnload moonraker.log and attach it here?  I would like to see what went wrong during the validation process.

JDWarner, 2022-10-14:
I suspect this may all have been related to how fluidd configures the settings in a different directory.  Here's the log.
[moonraker.log](https://github.com/Arksine/moonraker/files/9790681/moonraker.log)

JDWarner, 2022-10-14:
This was a totally fresh fluiddpi install, I moved to a bigger microSD and had dropped the printer config files in but no other changes.  Updated the packages, restarted the Pi and couldn't connect.

Eclypse3D, 2022-10-14:
Ok the system is back up but it is back to the other original issue of not being able to view gcode, or print (unable to open file) but I can access the configs now.

Arksine, 2022-10-14:
Ok, we can fix the gcode via ssh:
```
sudo service moonraker stop
cd ~/printer_data
ln -s ~/gcode_files gcodes
sudo service moonraker start
```

Arksine, 2022-10-14:
You may have some deprecation warnings as well.   You can use Mainsail to edit moonraker.conf and remove the deprecated options.

Eclypse3D, 2022-10-14:
yeah I got

 rmdir gcodes
rmdir: failed to remove 'gcodes': Directory not empty

Arksine, 2022-10-14:
What is in it?
`ls -l gcodes`

JDWarner, 2022-10-14:
Only one deprecation related to the secrets_path which is easily resolved.  But I also have 

`Failed to link subfolder 'database' to '/home/pi/.moonraker_database'. Folder 'database' already exists. This conflict requires manual resolution.`  

Things are working, should I just nuke one of these?  The one in the root of homedir I assume?

Eclypse3D, 2022-10-14:
looks like a link

lrwxrwxrwx 1 pi pi 20 Oct 14 19:28 gcode_files -> /home/pi/gcode_files

Arksine, 2022-10-14:
Ok, do the following:
```
sudo service moonraker stop
cd ~/printer_data
rm -rf gcodes
ln -s ~/gcode_files gcodes
sudo service moonraker start
```

Arksine, 2022-10-14:
@JDWarner Yes.  Remove `~/.moonraker_database` if it is a fresh install and doesn't contain items you need to recover.

Eclypse3D, 2022-10-14:
still unable to print, go to print, it uploads the file, then says unable to open file.  I check the gcode its in there, then a few seconds later the file disappears.

Arksine, 2022-10-14:
Ok.  Download the both klippy.log and moonraker.log and attach them.   If klippy.log is not there we may need to create that link as well.

Eclypse3D, 2022-10-14:
I can upload the file from just selecting upload from the gcode menu, and print but I cannot send a print to printer from Cura now, it just errors saying unable to open file.

JDWarner, 2022-10-14:
I also needed to link `~/gcode_files` into the new directory but from there everything appears to be working.  Uploaded a new gcode file and the preview works, no warnings.

Eclypse3D, 2022-10-14:
[moonraker (7).log](https://github.com/Arksine/moonraker/files/9790744/moonraker.7.log)
[klippylog.log](https://github.com/Arksine/moonraker/files/9790745/klippylog.log)

here's the goods

Eclypse3D, 2022-10-15:
Ok I got it working now..... I commented out the depreciation notices and gave a clean reboot, then went back into the gcodes menu in Mainsail and hit refresh its all in there now and test print from wifi is working.

MickyMellor, 2022-10-15:
If there are any others who installed via KIAUH for multiple printer set ups, here's what I can tell so far:

Assuming you used default naming, your instances will be numbered printer_1, printer_2 etc. I will use printer_X below for paths. You will need to repeat actions for each instance, replacing X with 1, 2, etc.

Your first 2 moonraker services will be using environment arguments from ~/printer_X_data/systemd/moonraker.env and the env file is still ok. It is providing the -d -c & -l arguments to the correct places.
If you have 3 or more instances, the 3rd onwards will have environment arguments in the service definition : /etc/systemd/system/moonraker-X.service
This is a much more involved fix and I found that updating the service to match instance 1 & 2 was probably the best way forward. 
So: 
cp printer_1_data/systemd/moonraker.env printer_X_data/systemd/moonraker.env
Edit the file to the correct X number in the -d -c & -l paths.

sudo mv /etc/systemd/system/moonraker-X.service /etc/systemd/system/moonraker-X.service.bak
sudo cp /etc/systemd/system/moonraker-1.service  /etc/systemd/system/moonraker-X.service

Edit the .service and change the path to printer_X_data/systemd/moonraker.env

Rinse and repeat for all instances then:
sudo systemctl daemon-reload

Then, for all instances we fix the folder structure:
cd ~/printer_X_data
mkdir database certs
ln -s ~/klipper_config/printer_X config
ln -s ~/klipper_logs logs
ln -s ~/gcode_files gcodes

Since I'm working with a blank install, I've created a new database folder rather than linking to ~/.moonraker-X_database
If you want separate gcode libraries for each instance, instead of linking to ~/gcode_files you could mkdir gcodes instead.
Works with mainsail, not sure how fluidd will react.

Deprecation warnings:
Edit ~/klipper_config/printer-X/moonraker.conf
remove:
enable_debug_logging: False
database_path: ...
config_path: ...
log_path: ...

Then:
sudo service moonraker-X restart

Arksine, 2022-10-15:
@MickyMellor The validator should be able to convert any number of instances assuming the naming scheme is consistent.  Could you upload `moonraker.log` for one of the instances that failed?  Thanks.

th33xitus, 2022-10-15:
I can confirm that it works for more than 2 instances. I just tested it with 4 in total. (ports 7125 to 7128)
But the password prompt always returned a message about a network error after submitting the sudo password.
[image]

Even though the error message at the bottom of the screenshot was shown, it seems like that the password was still accepted. Because after i restarted the corresponding moonraker service, Moonraker seemed to do its magic and create the symlinks.

So this was pretty much what i did:
- put in password for first instance
- get error message about network error
- ignore error message
- restart first instance service
- rinse and repeat for the remaining 3 services

MickyMellor, 2022-10-15:
> @MickyMellor The validator should be able to convert any number of instances assuming the naming scheme is consistent. Could you upload `moonraker.log` for one of the instances that failed? Thanks.

It seems that I am getting the same message as @th33xitus when entering the sudo password and a manual restart of the service brings it good.
It could also be that this is fresh install and so .moonraker_database_1 does not exist to be linked to, once the conf is updated to remove deprecated lines, all folders are created under printer_1_data
So config, gcodes and logs still *might need to be removed and sym linked to the correct location.

[moonraker-1.log](https://github.com/Arksine/moonraker/files/9792014/moonraker-1.log)
[moonraker-8.log](https://github.com/Arksine/moonraker/files/9792015/moonraker-8.log)

chopdi, 2022-10-15:
can someone show me his new config as an example?  no matter what I set, I can't upload anything from the slicer.

sosher, 2022-10-15:
Seems to me that mainsail and fluid are expecting to see the configurtation files in the new location, ~/printer_data/config. But in reality klipper and moonraker are using the /klipper_config

chopdi, 2022-10-15:
My hole sd Card is read only after the update... rly Strange

Arksine, 2022-10-15:
For everyone that experienced an issue running FluiddPi or MainsailOS I have created a script that should fix the datapath.  SSH into your pi and run the following:

```
cd ~/moonraker
git pull
./scripts/data-path-fix.sh
```

Note that this will only work on installs with the default configuration and a single instance.  The script will exit and report an error if the folders are not where it expects them to be.

Arksine, 2022-10-15:
FWIW, there is nothing that Moonraker does that can make the SD Card read-only.  That is an internal bit set via a low level register.

Trinity030, 2022-10-15:
any good way to fix moonraker

Trinity030, 2022-10-15:
> cd ~/moonraker
> git pull
> ./scripts/data-path-fix.sh

Error: unable to find database: /home/vz/.moonraker_database

Arksine, 2022-10-15:
@Trinity30

It appears you don't have a default installation on MainsailOS or FluiddPi.  Here are the contents of the script:

```bash
#!/bin/bash
# Data Path Fix for legacy MainsailOS and FluiddPi installations running
# a single instance of Moonraker with a default configuration

DATA_PATH="${HOME}/printer_data"
DATA_PATH_BKP="${HOME}/.broken_printer_data"
DB_PATH="${HOME}/.moonraker_database"
CONFIG_PATH="${HOME}/klipper_config"
LOG_PATH="${HOME}/klipper_logs"
GCODE_PATH="${HOME}/gcode_files"
MOOONRAKER_CONF="${CONFIG_PATH}/moonraker.conf"
MOONRAKER_LOG="${LOG_PATH}/moonraker.log"

[ ! -d "${DB_PATH}" ] && echo "Error: unable to find database: ${DB_PATH}" && exit 1
[ ! -f "${MOOONRAKER_CONF}" ] && echo "Error: unable to find config: ${MOOONRAKER_CONF}" && exit 1
[ ! -d "${LOG_PATH}" ] && echo "Error: unable to find log path: ${LOG_PATH}" && exit 1
[ ! -d "${GCODE_PATH}" ] && echo "Error: unable to find gcode path: ${GCODE_PATH}" && exit 1

sudo systemctl stop moonraker

[ -d "${DATA_PATH_BKP}" ] && rm -rf ${DATA_PATH_BKP}
[ -d "${DATA_PATH}" ] && echo "Moving broken datapath to ${DATA_PATH_BKP}" && mv ${DATA_PATH} ${DATA_PATH_BKP}

mkdir ${DATA_PATH}

echo "Creating symbolic links..."
ln -s ${DB_PATH} "$DATA_PATH/database"
ln -s ${GCODE_PATH} "$DATA_PATH/gcodes"
ln -s ${LOG_PATH} "$DATA_PATH/logs"
ln -s ${CONFIG_PATH} "$DATA_PATH/config"

~/moonraker-env/bin/python -mlmdb -e ${DB_PATH} -d moonraker edit --delete=validate_install

echo "Running install script"

~/moonraker/scripts/install-moonraker.sh -f -c ${MOOONRAKER_CONF} -l ${MOONRAKER_LOG}
```

Make a copy of the script and edit the following variables to match the correct location on your system:
```
DB_PATH="${HOME}/.moonraker_database"
CONFIG_PATH="${HOME}/klipper_config"
LOG_PATH="${HOME}/klipper_logs"
GCODE_PATH="${HOME}/gcode_files"
```

Given that you have a custom install I presume you are familiar with linux and bash?

ZerGo0, 2022-10-16:
Is there any fix for chroot users? This update seems to break a lot for me. Files can't be written anymore (upload doesn't work, existing files are read-only), service status doesn't work anymore, update status returns `INVALID` and more. I have to stay on edaeb17f389f89382256504228026562dc1bf8fe to use moonraker.

Trinity030, 2022-10-16:
> @Trinity030
> 
> It appears you don't have a default installation on MainsailOS or FluiddPi. Here are the contents of the script:
> 
> ```shell
> #!/bin/bash
> # Data Path Fix for legacy MainsailOS and FluiddPi installations running
> # a single instance of Moonraker with a default configuration
> 
> DATA_PATH="${HOME}/printer_data"
> DATA_PATH_BKP="${HOME}/.broken_printer_data"
> DB_PATH="${HOME}/.moonraker_database"
> CONFIG_PATH="${HOME}/klipper_config"
> LOG_PATH="${HOME}/klipper_logs"
> GCODE_PATH="${HOME}/gcode_files"
> MOOONRAKER_CONF="${CONFIG_PATH}/moonraker.conf"
> MOONRAKER_LOG="${LOG_PATH}/moonraker.log"
> 
> [ ! -d "${DB_PATH}" ] && echo "Error: unable to find database: ${DB_PATH}" && exit 1
> [ ! -f "${MOOONRAKER_CONF}" ] && echo "Error: unable to find config: ${MOOONRAKER_CONF}" && exit 1
> [ ! -d "${LOG_PATH}" ] && echo "Error: unable to find log path: ${LOG_PATH}" && exit 1
> [ ! -d "${GCODE_PATH}" ] && echo "Error: unable to find gcode path: ${GCODE_PATH}" && exit 1
> 
> sudo systemctl stop moonraker
> 
> [ -d "${DATA_PATH_BKP}" ] && rm -rf ${DATA_PATH_BKP}
> [ -d "${DATA_PATH}" ] && echo "Moving broken datapath to ${DATA_PATH_BKP}" && mv ${DATA_PATH} ${DATA_PATH_BKP}
> 
> mkdir ${DATA_PATH}
> 
> echo "Creating symbolic links..."
> ln -s ${DB_PATH} "$DATA_PATH/database"
> ln -s ${GCODE_PATH} "$DATA_PATH/gcodes"
> ln -s ${LOG_PATH} "$DATA_PATH/logs"
> ln -s ${CONFIG_PATH} "$DATA_PATH/config"
> 
> ~/moonraker-env/bin/python -mlmdb -e ${DB_PATH} -d moonraker edit --delete=validate_install
> 
> echo "Running install script"
> 
> ~/moonraker/scripts/install-moonraker.sh -f -c ${MOOONRAKER_CONF} -l ${MOONRAKER_LOG}
> ```
> 
> Make a copy of the script and edit the following variables to match the correct location on your system:
> 
> ```
> DB_PATH="${HOME}/.moonraker_database"
> CONFIG_PATH="${HOME}/klipper_config"
> LOG_PATH="${HOME}/klipper_logs"
> GCODE_PATH="${HOME}/gcode_files"
> ```
> 
> Given that you have a custom install I presume you are familiar with linux and bash?

i used kiauh for installing mailsailos   
:(

Arksine, 2022-10-16:
@ZerGo0 You may have to manually update your install and disable validation.  I don't run a chroot'ed install, so I am not familiar with what is required to make them work initially.  If I have time I'll see if I can get it working on an old android device, if you followed some specific instructions to get your current install working they would be useful.

I suspect you will need to do something like the following:

1) Stop the moonraker service
2) Create a primary data path as outlined in the [documentation](https://moonraker.readthedocs.io/en/latest/installation/#data-folder-structure).  You can either choose to move your existing folders to the datapath or create symbolic links.
3) Modify your service file to include the `-d <data_path>` argument, where `data_path` is the location of the data path
4) Remove deprecated options from your configuration.
5) Add the following to the `machine` section in your config to disable the validation process:

```
[machine]
validate_service: False
validate_config: False
```

Arksine, 2022-10-16:
@Trinity030 

Is this a new installation?

[64 more replies]