Arksine/moonraker #523: Moonraker after update stopped working

markomenino, 2022-10-15:
### What happened

After having updated the moonraker it was like in the image.

[image]

I already restarted the raspberry and ran the script: ~/moonraker/scripts/install-moonraker.sh
But it still doesn't open.
With the command sudo journalctl -xe -u moonraker.service in putty this appears:

[image]

### Client

Fluidd

### Browser

Chrome

### How to reproduce

When opening the fluid

user897943, 2022-10-15:
I was having errors on the notification bell, which I fixed. From what I can remember, I had to adjust the path to the virtual sd card to my home folder "/home/pi/..." instead of using the "~/...", remove the  old DB because there was a conflict of database folder already exists, and amend the config file to remove some deprecated stuff. Then it worked.

Arksine, 2022-10-15:
Moonraker has an internal process that corrects the service unit and the config, as mentioned in the announcement that went out last week.  To complete the process you have to enter your password...however if you attempted to reinstall with the install script it likely can't recover.  I have pushed a script that can be used to recover default installations on MainsailOS and FluiddPi.  SSH into the Pi and run the following:

```
cd ~/moonraker
git pull
./scripts/data_path_fix.sh
```

tbam, 2022-10-18:
> cd ~/moonraker
> git pull
> ./scripts/data_path_fix.sh

I believe this should read (dashes vs underscores in script name):
cd ~/moonraker
git pull
./scripts/data-path-fix.sh

Arksine, 2022-10-18:
You are correct, thanks tbam.

markomenino, 2022-10-19:
Problem solved.
Thanks.

closed 2022-10-19 by Arksine