Arksine/moonraker #528: Config Folder/Path - Possible Duplication?

nadavmatalon, 2022-10-16:
### What happened

Using 'Klipper-Moonraker-Mainsail' on Armbian Bullseye.

After the recent Config Folder/Path update in Moonraker, I can see my printer's config files (i.e printer.cfg, moonraker.conf, and others) in 2 locations: the originals under: '~/home/pi/klipper_config', but now also under '~/home/pi/printer_data/config'.

Could you please clarify if this should be the case (it seems logical that these files should be in only one location) and if not what should be done (without breaking the machine :-)

Thanks!

### Client

Other

### Browser

Chrome

### How to reproduce

Same printer config files can be found under 2 different folders:
~/home/pi/klipper_config
and under: 
'~/home/pi/printer_data/config

Arksine, 2022-10-16:
This is expected.  The latest update moves Moonraker to managed folder structure, where all files and subfolders are contained within one primary data folder, ie: `/home/pi/printer_data`.   To convert existing installations Moonraker creates symbolic links to the old folder locations in the data folder.  Thus they appear to be duplicated, but the files only exist in `/home/pi/klipper_config`.

New installations won't have these links, everything will be contained in the data folder.  Below is a visual of what a converted folder looks like:

[image]

You can see how the symbolic links point to the real location.

nadavmatalon, 2022-10-17:
> This is expected. The latest update moves Moonraker to managed folder structure, where all files and subfolders are contained within one primary data folder, ie: `/home/pi/printer_data`. To convert [...]

Thank you for taking the time to answer this and the detailed explanation! Makes sense.

On a (possibly) related note, I've been experiencing some strange behaviour with relation to editing and then uploading the files in this folder, specifically: I can edit and then upload the printer.cfg file without a problem, but trying to edit and upload any of the other files, including moonraker.conf, immediately raises an error ("unable to upload file" in Klipper). I doesn't seem to be a permissions issue as the printer.cfg file uploads fine. Also, I'm not sure if this issue comes from the Klipper side or that of Moonraker (or perhaps the interface between the two?). Any idea why that might be?

nadavmatalon, 2022-10-17:
Tried to reproduce just now and all went fine. Maybe the latest Moonraker update sorted it out or I just messed up somewhere. Sorry to have troubled you with this!

Arksine, 2022-10-17:
I have pushed a few fixes to handle edge cases, particularly those that occur on new installations via KIAUH.  Glad it is working now for you.

closed 2022-10-19 by Arksine