Arksine/moonraker #530: Can't update from 0.7.1-717 > 0.7.1.723 - see error logs

user897943, 2022-10-19:
### What happened

Press update on mainsail, this error comes up:
```
22:13:14	Updating Application moonraker...
22:13:15	Application moonraker: Updating Repo...
22:13:15	error: object file .git/objects/9a/466f3b224f18d6f91909b675c2db8d113573c9 is empty
22:13:15	error: object file .git/objects/9a/466f3b224f18d6f91909b675c2db8d113573c9 is empty
22:13:15	fatal: loose object 9a466f3b224f18d6f91909b675c2db8d113573c9 (stored in .git/objects/9a/466f3b224f18d6f91909b675c2db8d113573c9) is corrupt
22:13:15	fatal: the remote end hung up unexpectedly
22:13:16	Error updating moonraker
22:13:16	Error updating git repo
```
Just to make sure connectivity isn't the problem:
```
pi@mainsailos:~ $ sudo apt update
[sudo] password for pi:
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [392 kB]
Fetched 439 kB in 1s (428 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
pi@mainsailos:~ $
```

### Client

Mainsail

### Browser

Microsoft Edge

### How to reproduce

Click update on mainsail next to moonraker application

Arksine, 2022-10-19:
You have a corrupt repo and will need to do a hard recovery.  You can do this via SSH:
```
sudo service moonraker stop
cd ~
rm -rf moonraker
git clone https://github.com/arksine/moonraker.git
sudo service moonraker start
```

Keep in mind that this update may require your sudo password to complete.  After fixing the repo go to Mainsail and check your notifications.  If you see `Sudo Password Required` click the link and follow the instructions on the landing page.

user897943, 2022-10-20:
Thanks, fixed now

closed 2022-10-20 by user897943

Stooovie, 2022-11-05:
Why does this repo corruption thing occur? It happens all the time to me.

Arksine, 2022-11-05:
Could be a bad sdcard, however I believe the most common cause is removing power from from the host without properly shutting down.