Arksine/moonraker #535: Cannot upload the file!

Thomas-Austria, 2022-10-25:
### What happened

I can't upload or edit Gcode files or Printer.cfg

2022-10-23 12:13:44,515 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.100.7
2022-10-23 12:13:44,517 [app.py:log_request()] - 101 GET /websocket (192.168.100.7) [TRUSTED_USER] 3.93ms
2022-10-23 12:13:44,519 [websockets.py:open()] - Websocket Opened: ID: 547552341536, Proxied: False, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36, Host Name: 192.168.100.211
2022-10-23 12:13:44,530 [file_manager.py:get_file_list()] - Updating File List ...
2022-10-23 12:13:44,540 [app.py:log_request()] - 101 GET /websocket (192.168.100.7) [TRUSTED_USER] 2.04ms
2022-10-23 12:13:44,540 [websockets.py:open()] - Websocket Opened: ID: 547552083152, Proxied: False, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36, Host Name: 192.168.100.211
2022-10-23 12:13:44,546 [websockets.py:_handle_identify()] - Websocket 547552083152 Client Identified - Name: mainsail, Version: 2.3.1, Type: web
2022-10-23 12:13:44,629 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Metadata not available for <>
Traceback (most recent call last):
File "/home/pi/moonraker/moonraker/websockets.py", line 253, in execute_method
result = await callback(params)
File "/home/pi/moonraker/moonraker/websockets.py", line 362, in func
result = await callback(
File "/home/pi/moonraker/moonraker/components/file_manager/file_manager.py", line 332, in _handle_metadata_request
raise self.server.error(
utils.ServerError: Metadata not available for <>

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Upload & Print --> cannot upload the file!
Edit Printer.cfg --> Nothing happens 

### Additional information

Operating System of moonraker: Raspberry Pi OS Lite (64-bit) 2022-09-22

mainsail.cfg
[virtual_sdcard]
path: /home/pi/Ender3Pro_data/gcodes
on_error_gcode:
CANCEL_PRINT

printer.cfg
[virtual_sdcard]
path: /home/pi/Ender3Pro_data/gcodes

Arksine, 2022-10-25:
You will need to attach the full `moonraker.log`, the traceback you posted is not related to the error you are describing.

Thomas-Austria, 2022-10-25:
[moonraker.log](https://github.com/Arksine/moonraker/files/9863415/moonraker.log)

Arksine, 2022-10-25:
The only errors in the log are improperly formatted metadata requests.  There are no upload attempts logged, successful or otherwise.

I suspect that the error is on the browser side.  I would try hitting the "refresh directory" button for gcode and config files.  You may need to clear your cache.

Thomas-Austria, 2022-10-27:
The Problem was that Chrome blocks insecure private network requests. Disable **"Block insecure private network requests."** and it works fine.

closed 2022-10-27 by Thomas-Austria