Arksine/moonraker #536: Central Gcode File for multiple printer instances

thetailwind, 2022-10-26:
### What happened

After updating to the latest version of moonraker via Kiauh, i'm getting some errors:

`GCode path received from Klipper does not match expected location.

Received: '/home/XXX/gcode_files'
Expected: '/home/XXX/printer_1_data/gcodes'

Modify the [virtual_sdcard] section Klipper's configuration to correct this error.

[virtual_sdcard]
path: /home/XXX/printer_1_data/gcodes`

 While I understand this is by design for individual printers, it's not optimal for those running multiple instances. 
 
 What is the best approach to ensure there is one central folder to hold all the gcode files that is also available to all instances? 
 
 Thanks!

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Do a fresh install with multiple instances.

Arksine, 2022-10-26:
Use symbolic links.  If you have a shared`~/gcode_files` path, you can create a symbolic link to it at `~/printer_1_data/gcodes`, `~/pritner_2_data/gcodes`, etc. 

If you are modifying an existing installation make sure you stop the services first, as you will need to remove the existing `gcodes` folder in each data path.

Arksine, 2022-10-30:
I'm going to close this as discussion seems to have concluded.  Thanks.

closed 2022-10-30 by Arksine