Arksine/moonraker #537: extract_metadata() appears to be 10-20x slower than it should be

jschuh, 2022-10-30:
### What happened

After turning on `enable_object_processing` I started hitting gateway timeouts for uploading even medium sized gcode files (under 1mb), and larger files would hang for so long that I just gave up. Digging into it, I found this in `metadata.py`:

```
    for key in SUPPORTED_DATA:
        func = getattr(slicer, "parse_" + key)
        result = func()
        if result is not None:
            metadata[key] = result
    return metadata
```

It ends up iterating through a list of attributes that result in 10-20 method calls, most of which look something like this:

```
    def parse_filament_total(self) -> Optional[float]:
        return _regex_find_first(
            r"filament\sused\s\[mm\]\s=\s(\d+\.\d*)", self.footer_data)
```

For Prusa Slicer (and any Slic3r derivatives) all of the config comments are stored at the end of the file. So, that means running every one of these regexes individually over the entire gcode file on each pass. For the specific case of the config comments you could probably speed things up quite a bit by simply isolating the real footer from the body of the gcode. However, there are also a number of cases like this:

```
        matches = re.findall(
            r";BEFORE_LAYER_CHANGE\n(?:.*\n)?;(\d+\.?\d*)", self.footer_data)
```

Those really do need to scan all of the gcode, so even if you were to isolate out the relatively small footer you'd still end up with a bunch of redundant regex passes over the whole file.

So, it seems to me that even a relatively naive parser implementation that makes only a single pass would probably net a 10-20x speed up over the current implementation.

### Client

Other

### Browser

Other or N/A

### How to reproduce

Upload any sufficiently large gcode file with `enable_object_processing` turned on (in my case 1mb or more was enough to trigger on one machine).

jschuh, 2022-10-30:
And now that I've looked more closely I see how much of the time is spent in `preprocess_cancellation`, so maybe the overhead on extracting the metadata is just a drop in the bucket compared to that. 🤷

Arksine, 2022-10-30:
The metadata processor already limits searches to a section of the gcode file (the header or footer).  For single line items it could do the regex line by line rather than item by item, however it would probably be slower.  If someone wanted to test it they are free to do so.

As you have discovered, file processing is a different animal.  You are I/O bound on an SBC.  If you want to speed it up you should do it in the slicer.

jschuh, 2022-10-30:
> The metadata processor already limits searches to a section of the gcode file (the header or footer).

Not to quibble, but [everything after the first 512k is treated as the footer](https://github.com/Arksine/moonraker/blob/60b871adbe3ed363ca999d3a080cf335a71baddb/moonraker/components/file_manager/metadata.py#L955). Sorry if I oversimplified, but in my experience that means the "footer" is nearly the whole file.

> For single line items it could do the regex line by line rather than item by item, however it would probably be slower. If someone wants to test it they are free to do so and post their results, I would be open to making the change if the hard data justifies it.

I was more thinking a simple parser rather than just running every regex line by line. Either way, fair that there's no point in even proposing a change without a profile in hand demonstrating a real issue. Apologies on my part, I'll dig in a bit this week to see if there's enough overhead to even make a change worth it. If not I'll just close this bug out.

> As you have discovered, file processing is a different animal. You are I/O bound on an SBC. If you want to speed it up you should do it in the slicer.

I wouldn't say "discovered." My apprehension about text processing overhead comes from cutting my teeth on code to deal large text files on 1990s hardware. Todays SBCs feel over-resourced by comparison. 😉

I do wholeheartedly agree that best approach is to just have the slicer emit the necessary metadata. Case in point, I could quite easily write a PR for Prusa Slicer that emits the object boundary polygons as comment metadata. I'm familiar with the arrange code, which already generates the necessary outline, so it's just some plumbing to get it into the file. That stated, I'm not optimistic about getting it landed. The Prusa Slicer devs are polite enough, but of the PRs I've submitted to them, the ones they've accepted have mostly been either build system related or straightforward bug fixes.

Arksine, 2022-10-30:
> Not to quibble, but [everything after the first 512k is treated as the [...]

Prior to reading the footer there is a seek to `size - READ_SIZE`.  This moves the marker to 512 KiB before the end of the file before the read is performed.  The first 512k is the header, the last 512k is the footer.   If the file size is less than 1MiB there is some special handling.

> I was more thinking a simple parser rather than just running every regex line by line. Either way, fair that there's no point in even proposing a change without a profile in hand demonstrating a [...]

Sounds good.

> I do wholeheartedly agree that best approach is to just have the slicer emit the necessary metadata. Case in point, I could quite easily write a PR for Prusa Slicer that emits the object boundary [...]

The [preprocess-cancellation](https://github.com/kageurufu/preprocess_cancellation) module can be used with several slicers post-process, so no need to worry about a PR.  Maybe it something they will be interested in the future.

jschuh, 2022-10-30:
Yeah, I'm going to close this out. I just started running some profiling, and it's already clearly showing that nearly all of the time is spent in preprocess_cancellation, so there's no optimizations here that that would matter. Apologies for wasting your time on this. I shouldn't have filed the issue without running some profiles first.

> The preprocess-cancellation module can be used with several slicers post-process, so no need to worry about a PR. Maybe it something they will be interested in the future.

I know, and I already set that up, but for various reasons I'm uncomfortable with running unsigned 3p code on my local desktop. Anyway, I'm going to write that PR for PRusa Slicer and see if I can get it accepted. At a minimum I can run it in my custom build that I normally use anyway, since it includes my other patches.

Once again, sorry for filing an issue without doing that appropriate homework first (i.e. profiling). I get annoyed when I'm on the receiving end of that, and should know better than to do it to someone else.

closed 2022-10-30 by jschuh

jschuh, 2023-05-19:
FWIW, after this I wrote a patch for my own Prusa Slicer fork that generates the `exclude_object` commands, and I decided to [post it as a PR](https://github.com/prusa3d/PrusaSlicer/pull/10618) now that the upcoming Prusa Slicer adds a Klipper gcode flavor.