Arksine/moonraker #540: Latest update completely broke Fluidd UI

TxBillbr, 2022-11-03:
### What happened

After update process ran Fluid (up-to-date) could not find config or data files. (due to new filemanager handling) Apparently update was supposed to fix systemd service file but did not.  After manually configuring systemd per the documentation and verifying moonraker.env file, the restart was successful, but jobs lost all time/data information. No date/time for when printed and current time/date for last modified. So this may be a fluidd problem.

This issue may be closed as no action is required. 

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Unclear how to reproduce. One of the people I help has this exact condition and is waiting for me to "fix his printer". I can pull a copy of his logs if that would be of use. Reporting this in case for notification of possible problem.

[image]

### Additional information

A couple of folks that I help had the exact same issue, so not sure if the update code was supposed to make the changes or the user is expected to.  None of these folks are really competent to be on a linux system editing systemd service files.

Arksine, 2022-11-03:
See #516.  The update is automated, however modifying the service unit requires sudo permission on most systems.  Moonraker sends an announcement requesting sudo access, the user must click through and enter their linux password on the landing page.

If for some reason the update failed there is a `data-path-fix.sh` script that should work on stock MainsailOS and FluiddPi distros.  This script creates the appropriate symbolic links (database, configs, gcode path) then updates the service file.  Most likely you lost history/job info because a new database was created when you manually made the changes.

closed 2022-11-03 by Arksine

TxBillbr, 2022-11-03:
Arksine. Thanks for the quick reply. I had scanned the current issues but missed that one.

ahmedbadran732000, 2022-11-04:
Sudo Password Required

Moonraker requires sudo access to finish updating. Please click on the attached link and follow the instructions.
but it gave me after i put my password
Request failed with error: Unauthorized
[moonraker.log](https://github.com/Arksine/moonraker/files/9942333/moonraker.log)