Arksine/moonraker #541: Unable to connect MCU

AntoszHUN, 2022-11-04:
### What happened

Since yesterday's Moonraker update, I get this error and after a while the mcu 'mcu': Unable to connect error appears.

Is there any solution for this? What was changed again? Why wasn't it good the way it was?
[image]

### Client

Fluidd

### Browser

Firefox

### How to reproduce

I turn on the machine. The error comes automatically.

Arksine, 2022-11-04:
> Is there any solution for this? What was changed again? 

Its possible this is related to #516, however I can not diagnose unless you provide `moonraker.log`.  FWIW, mcu errors are unrelated to Moonraker.

> Why wasn't it good the way it was?

As of yet there is no stable release for Moonraker, the update manager pulls updates straight from the master branch of this repo.  This is developmental code.  Potential breaking changes are announced before they are pushed, however I do not know if KlipperScreen delivers these announcements to you.   Mainsail and Fluidd do.

Until Moonraker has a stable release I would recommend refraining from updating, or checking the pinned issues here before updating to see if they require any action.

AntoszHUN, 2022-11-04:
> Its possible this is related to #516, however I can not diagnose unless you provide `moonraker.log`. FWIW, mcu errors are unrelated to Moonraker.
>
>
> As of yet there is no stable release for [...]

All right. Thanks! I'll go after things then.

closed 2022-11-04 by AntoszHUN